pimoroni/pirate-audio #1: Mopidy config may contain wrong settings

kingosticks, 2019-11-19:
Your install procedures suggests running:
```
mopidy config | sudo tee /etc/mopidy/mopidy.conf
```
but this isn't ideal for two reasons:

1. The default config values for some settings depend on the user running Mopidy. i.e. running `mopidy config` gives different settings compared to running `sudo mopidyctl config`. Below shows the differences for the `core` section but other sections (e.g. `logging`, `m3u`, `local`) may be different also.
```
$ mopidy config
[core]
cache_dir = $XDG_CACHE_DIR/mopidy
config_dir = $XDG_CONFIG_DIR/mopidy
data_dir = $XDG_DATA_DIR/mopidy

$ sudo mopidyctl config
[sudo] password for pi: 
Running "/usr/bin/mopidy --config /usr/share/mopidy/conf.d:/etc/mopidy/mopidy.conf config" as user mopidy
[core]
cache_dir = /var/cache/mopidy
config_dir = /etc/mopidy
data_dir = /var/lib/mopidy
```
If you really must do this (see #2 for why not) then instead use:
```
sudo mopidyctl config | sudo tee /etc/mopidy/mopidy.conf
```

2. You'll need to be more careful when updating Mopidy and check the old default values you've specified are still suitable with the new version you just upgrade to. 

The [Mopidy documentation](https://docs.mopidy.com/en/latest/config/) suggests:
> When you have created the configuration file, open it in a text editor, and add the config values you want to change. If you want to keep the default for a config value, you should not add it to the config file, but leave it out so that when we change the default value in a future version, you won’t have to change your configuration accordingly.

Gadgetoid, 2019-11-20:
Thanks, this is some really useful insight. I've applied the `mopidyctrl` based "fix" for the time being, but I think in light of your point 2 this only constitutes a temporary fix.

This clued me into the existence of `/usr/share/mopidy/conf.d` though, and I wonder if that's perhaps a better place to drop plugin-specific config, and setup-specific settings so that they can be maintained independent of the default settings?

kingosticks, 2019-11-21:
I wanted to use that folder a while back for pimusicbox but I think I was talked out of it. Sadly I can't remember the exact reason but I think we might have been considering removing it at the time - people already struggle with the different config files (service vs user) so having more isn't helpful. Maybe @jodal remembers.

In general, /etc/mopidy/mopidy.conf is the right place for a user's specific config *overrides*, we just like to avoid redefining the entire config again. I think having those 3-4 specific settings you want to change there would be fine, and a bit clearer when people come to upgrade.

jodal, 2019-11-21:
`/usr/share/mopidy/conf.d` is a directory created by the `mopidy` Debian package for extension Debian packages to install defaults into, in case they need different defaults when installed as a Debian package than what they ship on PyPI.

`/etc/mopidy/mopidy.conf` is for the user's specific config overrides. The less you put in here, the easier it becomes to upgrade Mopidy in the future as you automatically get updated sane default values for all config values you don't specify.

So, use `/etc/mopidy/mopidy.conf` for your changes and use `sudo mopidyctl config` to see the total picture from the perspective of the Mopidy system service.

Gadgetoid, 2019-11-21:
Thank you for weighing in here- it's greatly appreciated. It looks like the path forward is clear, and I'll aim to echo out a basic starter config into `/etc/mopidy/mopidy.conf` rather than using Mopidy to build out a complete list of config info.

sandyjmacdonald, 2019-11-21:
Would you be interested in playing with one of the new Pirate Audio boards, @kingosticks ? No strings attached, just pop us an email with a shipping address and we can sort it out.

kingosticks, 2019-11-21:
Thanks but mine came in the post this morning! Will play with it this
weekend.

sandyjmacdonald, 2019-11-21:
🙂

jodal, 2019-11-21:
If you ship to Norway, I'd love one! My 7 year old has already begged me to get him one of these music devices running the software dad spends his nights hacking on for Christmas ;-)

sandyjmacdonald, 2019-11-21:
@jodal Pop an email to sandy at pimoroni dot com and I'll sort it 🙂

Gadgetoid, 2019-11-21:
In a bid to make our install script less terrible it now backs up `/etc/mopidy/mopidy.conf` and applies a default configuration for `raspberry-gpio` and `pidi` including some additional minimal options for `mpd`, `http`, `audio` and `spotify`.

kingosticks, 2019-11-22:
Looks good. I left a comment at https://github.com/pimoroni/pirate-audio/commit/a5740728dd7d12717b7759b442035a33b16a5eaa#r36074921

kingosticks, 2019-11-22:
Ahh ok, not quite there. The problem with this is that the Mopidy Debian package ships some service-specific config vales (below) and now these are lost. These are some of the ones getting the wrong values with the original implementation.

```
[core]
cache_dir = /var/cache/mopidy
config_dir = /etc/mopidy
data_dir = /var/lib/mopidy

[logging]
config_file = /etc/mopidy/logging.conf
debug_file = /var/log/mopidy/mopidy-debug.log

[local]
media_dir = /var/lib/mopidy/media

[m3u]
playlists_dir = /var/lib/mopidy/playlists
```
I think you could argue that the above technically belongs in `/usr/share/mopidy/conf.d`...

Gadgetoid, 2019-11-25:
Thank you. I've addressed your comment (ooh that was a silly mistake, in my defence I've been *really* ill over the last couple of weeks) and changed the config to be appended rather than simply overwriting `mopidy.conf`.

Looks like I need to re-image and see what I end up with now.

Furthermore some way to detect an old botched config and repair it might be handy- I guess a `sudo apt install --reinstall mopidy` would be a good start.

kingosticks, 2019-11-25:
I don't think that apt command will touch any *existing* config files, they are treated as a special case in the Deb package world. You'd want to first manually remove the config file, or remove the package (with `--purge`) first.

But how to detect the config file is botched? We do out best to validate and emit log messages when a value is clearly wrong but it's otherwise hard to tell.

Gadgetoid, 2019-11-26:
Turns out the secret sauce for fixing (read: eliminating and replacing with the default config) a previously botched config file is:

```
sudo apt install --reinstall -o Dpkg::Options::="--force-confask,confnew,confmiss" mopidy
```

For existing users with possibly botched configs, it might be worth just giving them some manual steps to fix everything up.

kingosticks, 2019-11-26:
Im surprised that deleting the config file and then doing a normal reinstall doesn't fix it. But whatever works.

commit a5740728 (2019-11-21): Don't fill mopidy.conf with default values, fixes #1