pimoroni/pirate-audio #3: Iris needs sudo access

sandyjmacdonald, 2019-11-20:
One way (albeit a little dubious) to solve this is: `sudo echo "mopidy ALL=NOPASSWD: /usr/local/lib/python2.7/dist-packages/mopidy_iris/system.sh" >> /etc/sudoers`

Gadgetoid, 2019-11-20:
Bizarre- this is documented in the install guide here: https://github.com/jaedb/Iris/wiki/Getting-started

Where it suggests:

> Run `sudo echo "mopidy ALL=NOPASSWD: /usr/local/lib/python2.7/dist-packages/mopidy_iris/system.sh" >> /etc/sudoers` to allow mopidy user sudo permission to run the Iris installer. If you've customised your install location, make sure you change the path to Iris' `system.sh` file.

But I never ran into this error, or indeed any kind of install process- at what point did you encounter it?

Looking at the script itself, it doesn't seem to ... do anything: https://github.com/jaedb/Iris/blob/master/mopidy_iris/system.sh

Gadgetoid, 2019-11-20:
Aha- it's used to back some of the buttons in the Iris settings UI. Managed to replicate. I'll add this line to the installer script.

closed 2019-11-21 by Gadgetoid