pimoroni/pirate-audio #5: Mopidy config should disable RaspPi on-board sound

AuntDaisy, 2019-11-25:
Two users have reported "no sound" issues with Pirate Audio and default install script.

wildestpixel suggested this fix - confirmed to work by at least one user.
In /boot/config.txt  
`#dtparam=audio=on`

See :
https://forums.pimoroni.com/t/no-sound-from-pirate-audio-smol-speaker/12223
https://forums.pimoroni.com/t/no-sound-from-pirate-audio-3w-stereo-amp/12227

Gadgetoid, 2019-11-25:
Oof, I wanted to avoid being opinionated about disabling onboard audio since it's possible to route sound properly without needing to do so- but possible != easy in this case.

Thank you for clarifying this. I agree. I'll ensure the installer disables onboard audio.

AuntDaisy, 2019-11-25:
Dear Philip,

 

                Thanks for that, it will be very helpful for users like me.

 

                I was really excited to get the Pirate Audio 3W Stereo Amp on Friday and spent Saturday morning installing Raspbian and the installer – no sound. Tried again with two different Raspbians and lots of fiddling – no sound.

I’d even read the README.md and not twigged what the problem was – it was only after wildestpixel’s helpful suggestions on the forum that it worked.

 

                I’m an experienced Delphi / C# programmer (scientific software / hardware) who’s new-ish to RaspberryPi, so I suspect other people will have similar problems…

 

                Thanks for your help,

 

                Austin

Gadgetoid, 2019-11-26:
I've implemented what should hopefully be a better fix- specifically changing `mopidy.conf` to use `output = alsasink device=hw:sndrpihifiberry` to specify which output device to use. This will presumably let users continue to use onboard audio for other things.

el1-x, 2020-10-01:
I have just put together a RPi Zero W with the Priate Audio line out pHat and I experienced the same issue mentioned above.
I had to set #dtparam=audio=off in /boot/config.txt to have everything working.

arcstar7, 2021-12-10:
> Dear Philip, Thanks for that, it will be very helpful for users like me. I was really excited to get the Pirate Audio 3W Stereo Amp on Friday and spent Saturday morning installing Raspbian and the [...]

This is a fairly late response, but wanted to indicate that I had the exact same issue.  Until finding the thread about disabling onboard audio, my speakers were not working as well.