pimoroni/pirate-audio #9: mopidy 3.x breaks installation

dkinon, 2019-12-29:
On December 22, 2019 (6 days ago) mopidy released version 3.0.0 which moves mopidy from python2 to python3.  This move breaks the installer since it is designed for python2.  Trying to move all mopidy dependencies to python3 is not currently possible due to a lack of python3 support for the Mopidy-PiDi and mopidy-raspberry-gpio python packages.  The only way I could get this project/installer to work was to modify the installer to force the latest 2.x version of mopidy:
`apt install -y --allow-downgrades mopidy=2.3.1-1 mopidy-spotify`

Gadgetoid, 2019-12-29:
Migrating our Mopidy packages to Python 3/Mopidy 3.0.0 is on my TODO list.

Mopidy Raspberry GPIO at least has a pending PR: https://github.com/pimoroni/mopidy-raspberry-gpio/pull/3

Gadgetoid, 2019-12-29:
@kingosticks looking into getting everything updated, tested and pushed to support Mopidy 3.x.x. However I've hit a roadblock I updated to Mopidy 3.0.1 and configuration seems to have broken. Mopidy sees the following config:

```
pi@raspberrypi:~/pirate-audio/mopidy $ sudo mopidyctl config
Running "/usr/bin/mopidy --config /usr/share/mopidy/conf.d:/etc/mopidy/mopidy.conf config" as user mopidy
[core]
cache_dir = /var/cache/mopidy
config_dir = /etc/mopidy
data_dir = /var/lib/mopidy
max_tracklist_length = 10000
restore_state = false

[logging]
verbosity = 0
format = %(levelname)-8s [%(threadName)s] %(name)s %(message)s
color = false
config_file =

[audio]
mixer = software
mixer_volume = 40
output = alsasink device=hw:sndrpihifiberry
buffer_time =

[proxy]
scheme =
hostname =
port =
username =
password =

[file]
enabled = true
media_dirs =
  $XDG_MUSIC_DIR|Music
  ~/|Home
excluded_file_extensions =
  .directory
  .html
  .jpeg
  .jpg
  .log
  .nfo
  .pdf
  .png
  .txt
  .zip
show_dotfiles = false
follow_symlinks = false
metadata_timeout = 1000

[http]
enabled = false  ; Extension disabled due to config errors.
hostname =   ; Must be a resolveable hostname or valid ip
port = 6680
zeroconf = Mopidy HTTP server on $hostname
allowed_origins =
csrf_protection = true
default_app = mopidy

[m3u]
enabled = true
base_dir =
default_encoding = latin-1
default_extension = .m3u8
playlists_dir =

[softwaremixer]
enabled = true

[stream]
enabled = true
protocols =
  http
  https
  mms
  rtmp
  rtmps
  rtsp
metadata_blacklist =
timeout = 5000
```

Yet the config file reads:

```
pi@raspberrypi:~/pirate-audio/mopidy $ sudo !!
sudo cat /etc/mopidy/mopidy.conf
# For information about configuration values that can be set in this file see:
#
#   https://docs.mopidy.com/en/latest/config/
#
# Run `sudo mopidyctl config` to see the current effective config, based on
# both defaults and this configuration file.

[raspberry-gpio]
enabled = true
bcm5 = play_pause,active_low,250
bcm6 = volume_down,active_low,250
bcm16 = next,active_low,250
bcm20 = volume_up,active_low,250

[pidi]
enabled = true
display = st7789

[mpd]
hostname = 0.0.0.0 ; Allow access to mpd from other devices

[http]
hostname = 0.0.0.0 ; Allow access to HTTP/Iris from other devices

[audio]
mixer_volume = 40
output = alsasink device=hw:sndrpihifiberry

[spotify]
enabled = false
username =       ; Must be set.
password =       ; Must be set.
client_id =      ; Must be set.
client_secret =  ; Must be set.
```

And, indeed, no hostname is seen when starting Mopidy:

```
pi@raspberrypi:~/pirate-audio/mopidy $ sudo -u mopidy /usr/bin/mopidy --config /etc/mopidy/mopidy.conf
INFO     2019-12-29 17:50:38,626 [10618:MainThread] mopidy.__main__
  Starting Mopidy 3.0.1
INFO     2019-12-29 17:50:38,727 [10618:MainThread] mopidy.config
  Loading config from builtin defaults
INFO     2019-12-29 17:50:38,732 [10618:MainThread] mopidy.config
  Loading config from file:///etc/mopidy/mopidy.conf
INFO     2019-12-29 17:50:38,735 [10618:MainThread] mopidy.config
  Loading config from command line options
INFO     2019-12-29 17:50:39,079 [10618:MainThread] mopidy.__main__
  Enabled extensions: file, softwaremixer, m3u, stream
INFO     2019-12-29 17:50:39,079 [10618:MainThread] mopidy.__main__
  Disabled extensions: http
WARNING  2019-12-29 17:50:39,080 [10618:MainThread] mopidy.__main__
  Found http configuration errors. The extension has been automatically disabled:
WARNING  2019-12-29 17:50:39,080 [10618:MainThread] mopidy.__main__
    http/hostname must be a resolveable hostname or valid IP
WARNING  2019-12-29 17:50:39,081 [10618:MainThread] mopidy.__main__
  Please fix the extension configuration errors or disable the extensions to silence these messages.
```

Am I being stupid?

dkinon, 2019-12-29:
@Gadgetoid not sure if this answers your question or not but here is the working config I was using for mopidy 3.0.1 before I decided to downgrade:
`# For information about configuration values that can be set in this file see:
#
#   https://docs.mopidy.com/en/latest/config/
#
# Run `sudo mopidyctl config` to see the current effective config, based on
# both defaults and this configuration file.

[raspberry-gpio]
enabled = true
bcm5 = play_pause,active_low,250
bcm6 = volume_down,active_low,250
bcm16 = next,active_low,250
bcm20 = volume_up,active_low,250

[pidi]
enabled = true
display = st7789

[mpd]
enabled = true
hostname = 0.0.0.0

[http]
enabled = true
hostname = 0.0.0.0
port = 6680
zeroconf = Mopidy HTTP server on $hostname
allowed_origins =
csrf_protection = true
default_app = mopidy

[audio]
mixer_volume = 40
output = alsasink device=hw:sndrpihifiberry

[spotify]
enabled = true
username = XXXXXXXXXXXXXXXX
password = XXXXXXXXXXXXXXXX
client_id = XXXXXXXXXXXXXXXX
client_secret = XXXXXXXXXXXXXXXX

[iris]
country = US
locale = en_US`

This may be a bug in mopidy 3.x but it would seem that `http/hostname` doesn't seem to like the trailing comment `; Allow access to HTTP/Iris from other devices`.

dkinon, 2019-12-29:
also @Gadgetoid if you would like me to pull together a PR with the changes I made to the installation script to lock mopidy package to 2.3.1-1, let me know.

kingosticks, 2019-12-29:
So, is it the comment for the http/hostname that breaks it? Also, looks like you need to install Mopidy-Mpd.

docmollo, 2019-12-29:
I was working on this issue with @dkinon yesterday, and I definitely made progress in getting things working with 3.x by removing the mid-line comment on the http hostname.

Gadgetoid, 2019-12-30:
@dkinon that might be a good short-term fix while I get everything migrated. A PR would be appreciated if you have the time, thank you!

Gadgetoid, 2019-12-30:
Can confirm it's the comment on the end of the http/hostname that breaks it. Is this something that warrants an upstream bug report?

rCarto, 2019-12-30:
Hello, if I want to set up a pirate audio device with mopidy today what should I do?

docmollo, 2019-12-30:
@rCarto Sorry for the delay....I didn't have my Pirate Radio setup to test the work we did over the weekend, so I had to do that first. Good thing, too - it turns out the Spotify plugin, which the install.sh script installs for you, also needs to be pinned to a previous version.

So to get this working today, here are the steps I took this morning to get Spotify working.  If you don't want to use Spotify, you can check out the other [Mopidy extensions](https://mopidy.com/ext/) and set things up to your own liking after step 6 below.

1. Get Raspbian Buster setup on your Pi to your liking (network, raspi-config options etc)
1. ```git clone https://github.com/pimoroni/pirate-audio```
1. ```cd pirate-audio/mopidy```
1. Use your favorite editor to edit the install.sh
   1. At line 65, change/delete/comment out the apt line and replace it with:
   ```
   apt install -y --allow-downgrades mopidy=2.3.1-1 mopidy-spotify=3.1.0-0mopidy1
   apt-mark hold mopidy mopidy-spotify
   ```
1. Save the file
1. ```sudo ./install.sh``` (this took a while on my Pi Zero...be patient)
1. Edit ```/etc/mopidy/mopidy.conf``` and put your Spotify username and password in the section at the bottom of the file
1. Visit [Mopidy-Spotify](https://mopidy.com/ext/spotify/) and scroll down to the Authentication section and click the green Authenticate button to allow Mopidy access to your Spotify account
1. Copy the two lines that are created into your mopidy.conf file (there are already blank placeholders in the conf file, so delete those, and copy the generated ones, or simply copy the values over).
1. Reboot the Pi

After the Pi boots, the LCD should have the UI up on it with a web URL to visit to control Mopidy. The URL takes you to the main Mopidy interface on the Pi, so to get the Spotify controls, click the Iris web client link on the page.

rCarto, 2019-12-30:
Thank you very much @docmollo !

Gadgetoid, 2019-12-31:
Thank you for weighing in on this @docmollo - it's very appreciated!

I will be able to give the Python 3 ports the attention they deserve when I'm back in the office- alas our Pirate Audio launch came at a somewhat awkward time!

docmollo, 2019-12-31:
You're welcome, @Gadgetoid! Heh...yeah, you guys have had a pretty insanely busy few months...throw in the holidays and I am surprised your brains haven't haven't melted out of your ears yet. ;) Hopefully when you're all back on the 6th things go smoothly!

Gadgetoid, 2019-12-31:
Still plugging away at 3.x.x support and running into more issues making it difficult to test & release. In particular, and what puts my mind at least somewhat at ease -- and in strong favour of version pinning the Python 2.x-compatible releases -- is that Iris does not yet support Mopidy/Python 3.x: https://github.com/jaedb/Iris/issues/356

In the mean time I've started a port of mopidy-pidi to Python 3.

dkinon, 2020-01-01:
@Gadgetoid @docmollo here is the PR I promised and Happy New Year!

Gadgetoid, 2020-01-02:
@dkinon you're a star! I'll give it a go now.

Rocketansky, 2020-01-04:
I tried an install with the latest install.sh. It includes the modification mentioned by @docmollo above.
I end up with the following errors:
Installing mopidy packages
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Version '2.3.1-1' for 'mopidy' ws not found
E: Unable to locate package mopidy-spotify
mopidy was already set on hold.
E: Unable to locate package mopidy-spotify

I have no knowledge about mopidy - any help is appreciated.

exviia, 2020-01-04:
Thanks a lot docmollo!!

docmollo, 2020-01-04:
@exviia you're welcome....but it was a team effort! Hopefully you got things working.

@Rocketansky Hmmm....that sounds like your apt sources aren't set up correctly. I haven't tried the install since the changes were merged into the Pimoroni source, but I just checked and the install.sh file that I tested with is exactly like a freshly cloned copy. Steps 4 and 5 in my comment above are now obsolete - you shouldn't need to make any changes to a freshly cloned install.sh file.

You could try checking the sha256 checksum of the install.sh file - it should look like this:

`pi@piaudio:~/temp/pirate-audio/mopidy $ sha256sum install.sh 
d9ad7b1c7ed5c0a266407e6202c69c7fd62cab0ad8354507c0a65193df5c46f1  install.sh
`

If the output of sha256sum doesn't match the above, I would go up a few directories, blow away your cloned copy of the code and do the git clone again.

I'll edit my comment above to indicate things are now fixed and add a Tweet to my thread with that info, too. Let me know how it goes.

Rocketansky, 2020-01-04:
Hi Dustin!
Thanks for the quick reply.
I deleted the whole directory and cloned it again.
The result of install.sh is the same as mentioned above.
Then i verified the checksum as you described. Got the same value.

You mentioned a missing apt source. Is it possible that you added one manually that is not in the script?
--
Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

dkinon, 2020-01-05:
@Rocketansky we specifically covered that case in the script so I believe there is something else going on there.

You can see in the script that you should end up with /etc/apt/sources.list.d/mopidy.list, does that file exist for you?

Please send output from the following commands:
- apt-cache show mopidy | grep ^Version
- lsb_release -a

Also please send updated output from the script so we can review it in detail.

kingosticks, 2020-01-05:
Since you only finding mopidy 2.2.3-1 (the latest currently available in Debian's repo) and not finding mopidy-spotify at all sounds exactly like you do not have `apt.mopidy.com` added to your package sources. On top of the good suggestions by @dkinon, you can also run `apt-cache policy | grep Packages` to get a list of what sources you have.

Rocketansky, 2020-01-05:
@kingosticks maybe that is the reason because the script searches for 2.3.1-1.
I don't know which one is correct.

Rocketansky, 2020-01-05:
@dkinon the file mopidy.list exists in the mentioned directory but has 0 bytes size.
The output of "apt-cache show mopidy | grep ^Version" is 2.2.2-1
The output of "lsb_release -a" is:
```
No LSB modules are available.
Distributor ID: Raspian
Description:     Raspian GBU/Linux 10 (buster)
Release:           10
Codename:      buster
```
Here the requested output of install.sh: 
[install_sh.txt](https://github.com/pimoroni/pirate-audio/files/4023604/install_sh.txt)

docmollo, 2020-01-05:
@kingosticks we're specifically installing version [2.3.1-1 of mopidy](https://github.com/pimoroni/pirate-audio/blob/master/mopidy/install.sh#L9) and [3.1.0-0mopidy1 of mopidy-spotify](https://github.com/pimoroni/pirate-audio/blob/master/mopidy/install.sh#L10) from the mopidy apt repo due to the switch to python 3 in December.

@Rocketansky The mopidy sources file being zero bytes is your problem. You'll need to either figure out what's going on with your system that's causing that file to not be written by the install script at [line 85](https://github.com/pimoroni/pirate-audio/blob/master/mopidy/install.sh#L85), or start with a fresh install of raspbian buster.

kingosticks, 2020-01-05:
Very sorry, I was in la-la land and wrote a complete load of garbage earlier. What I should have said is *since you are not finding mopidy-spotify at all (it's not available in Debian's repo)*, that sounds exactly like you do not have `apt.mopidy.com` added to your package sources. Which is confirmed since the `mopidy.list` file in question is zero bytes long. 

To be totally clear, @Rocketansky, the version in the script is correct. You should remove `/etc/apt/sources.list.d/mopidy.list` and rerun the script.

docmollo, 2020-01-05:
@kingosticks haha no worries! I am considering myself still on Holiday Time until I go back to work on Monday morning, so just chalk it up to that, I say! I'm surprised I'm as lucid in this thread as I am given that ;)

Rocketansky, 2020-01-05:
@kingosticks thx for the hint to delete the mopidy.list. That brought me one step further. But still the script did not finish successfully.
I followed then the advice of @docmollo to start from scratch.
Before the installation attempt I extended the partition to use the whole SD card. After that the installation script worked well.
Due to the experience of the last days I consider that script as very FRAGILE.
A easy way to improve the installation would be to declare minimum space requirements.

docmollo, 2020-01-06:
@Rocketansky congratulations - we're happy you got your setup working! And thank you for the feedback - we'll take it under advisement. Take care.

kingosticks, 2020-01-08:
> Is this something that warrants an upstream bug report?

Yes, I've finally create a new issue for it. Sorry for the delay in doing so.

bradmason25, 2020-01-21:
Hey,
I'm having very similar issues to Rocketansky with this error during installation
`E: Version '2.3.1-1' for 'mopidy' was not found
E: Version '3.1.0-0mopidy1' for 'mopidy-spotify' was not found`

I've read through your comments helping them and my mopidy.list file is correctly filled and updating.
I can't say I understand their issue with the partition size but I've tried several OS rewrites today with a fresh Raspian Buster (Pi0w) on a new 128GB SD and a 16GB SD that usually works fine

Any ideas?

aajshaw, 2020-01-21:
Just got the very same problem as bradmason25, fresh install of of Raspian Buster on a Pi zero W using a 16GB SD. Are there any diagnostics that I can provide that can help?

dkinon, 2020-01-21:
@bradmason25 @aajshaw happy to help with troubleshooting.  If apt isn't finding the specific version of the packages we need then my expectation is that either the mopody repo didn't get setup correctly or apt's cache didn't successfully update between the time that the repo was setup and the package install requested.  To get more information on the problem at hand, please provide the following:

The Mopidy apt source contents:
```
$ cat /etc/apt/sources.list.d/mopidy.list
# Mopidy APT archive
# Built on Debian 10 (buster), compatible with Ubuntu 19.10 and newer
deb http://apt.mopidy.com/ buster main contrib non-free
deb-src http://apt.mopidy.com/ buster main contrib non-free
```

The output of apt-cache show for the mopidy package:
```
$ apt-cache show mopidy | grep Version
Version: 3.0.1-2
Version: 3.0.1-1
Version: 2.3.1-1
Version: 2.3.0-1
Version: 2.2.3-1
...
Version: 0.19.5-1
```
The assumption is that the above will be empty since we shouldn't have any candidates. If it is empty, then please run `apt-get update` and run the same command again.

jodal, 2020-01-21:
I believe this new error is caused the combination of the Pirate Audio scripts/instructions installing specific versions from APT, which isn't a normal thing to do from APT, and that the Mopidy project just changed our APT repo setup from using a primitive repo builder called `apt-ftparchive`, that never cleans out old versions, to a more advanced repo builder called `reprepro`, which automatically removes all but the latest version from each distribution. After this change, the only version of Mopidy available from the `buster` distribution on apt.mopidy.com is the latest, 3.0.1-2, which requires Python 3.

I think the best way forward for Pirate Audio is to either upgrade to Python 3 and Mopidy 3 (obviously the best approach IMHO), or to use the versions of Mopidy shipped with Raspbian. apt.mopidy.com is intended for people that want to be more on the bleeding edge than the Debian distributions usually are. Currently, Pirate Audio is not up to date enough to work with the bleeding edge, and should thus stay with the Debian versions.

bradmason25, 2020-01-21:
@dkinon thanks for offering to help!
Results below:
```
pi@raspberrypi:~ $ cat /etc/apt/sources.list.d/mopidy.list
# Mopidy APT archive
# Built on Debian 10 (buster), compatible with Ubuntu 19.10 and newer
deb https://apt.mopidy.com/ buster main contrib non-free
deb-src https://apt.mopidy.com/ buster main contrib non-free
```
and
```
pi@raspberrypi:~ $ apt-cache show mopidy | grep Version
Version: 3.0.1-2
Version: 2.2.2-1
```

jodal, 2020-01-21:
> Version: 3.0.1-2

This is from apt.mopidy.com's buster distribution.

> Version: 2.2.2-1

This is from Raspbian buster.

dkinon, 2020-01-21:
@jodal sure enough, if I do an `apt-get update` now, all the previous versions are stripped away so just like mopidy upgrading to python3 in 3.x, one day what we have works and the next day it doesn't.  Any silliness we are doing to hold package versions is to bandaid the installation process until we can upgrade all the downstream components to python3.

@Gadgetoid at this point I'm willing to ditch the mopidy repo all together, at least until we are ready with python3 support.  Is there any reason this project wouldn't be perfectly fine with mopidy version `2.2.2-1` provided by raspbian buster?  It might seem a little extreme, but I'd like to get this to a place where the landscape stops shifting under us.  And as @jodal points out, bleeding edge and not stability is the intent of the hosted mopidy apt repo.

Gadgetoid, 2020-01-22:
@jodal I appreciate your need to move forward and avoid maintaining multiple different codebases and versions, but without a full suite of plugins/add-ons compatible with Python 3.x it's not possible for us to keep up (although I think we're really only waiting on Iris and our own plugins at the moment?). Is it possible that you could ship the `2.3.1-1` release of Mopidy alongside the 3.x distribution or are you wary of the support burden it might impose?

@dkinon I think extreme may be the only choice we have- the other alternative being to, perhaps, ship individual .deb files representing a complete Mopidy install post the Python 3 upgrade.

My professional experience suggests Mopidy shipping `2.3.1-1` via their own repos would be preferable to using `2.2.2-1` which may impose *more* upstream support burden. I really appreciate the *extremely* hands-on help and support we've got here from the Mopidy team and I'm loathe to do anything that would cause them more hassle than it's worth.

A little voice in my head is saying "forge ahead with Python 3.x and hope Iris catches up soon", but I'm a little afraid to go that route because- to me at least- Iris is *the* front-end for Mopidy (I always thought it was a first-party add-on!) and I really am at a loss as to how to find/select music otherwise.

kingosticks, 2020-01-22:
It sounds like Iris is very nearly ready for a Python 3 release. That being said, the other two existing Python 3 compatible webclients do certainly support selecting music. But I do admit that musicbox-webclient isn't half as slick as Iris

With that mind, can we help anywhere with porting the Pirate Audio extensions? Is there a work in progress branch to continue from? Is there any questions or specific issues we can help with? I take it you're happy to workaround that config inline comment regression.

dkinon, 2020-01-22:
@Gadgetoid if you don't have a problem with me storing the required debs for mopidy `2.3.1-1` in the repo and removing the mopidy repo from the installer in favor of straight deb installs, I'll do that today.

This only needs to work until we get the remaining bits of python3 support figured out and then we can decide whether to go back to the mopidy repo (which is admittedly bleeding edge) or go with some other solution to install mopidy.

Gadgetoid, 2020-01-23:
@dkinon I might be able to host the deb files on our own domain, so at least we wont gum GitHub up with binary files- but yes, something along those lines I think is necessary in the short term or we're going to have people bouncing off this installer and all the mayhem that entails.

@kingosticks thank you- I think I've got everything basically ready for Mopidy 3.x / Python 3.x, but lack of Iris for Mopidy 3.x meant I couldn't finish testing and satisfy myself they were ready to ship. (and couldn't update this installer, since we rely heavily on Iris to provide the UI)

- Mopidy-PiDi - https://github.com/pimoroni/mopidy-pidi/pull/7
- PiDi plugins - https://github.com/pimoroni/pidi-plugins (which should all work on 2/3)
- Mopidy Raspberry GPIO - https://github.com/pimoroni/mopidy-raspberry-gpio/pull/3

Gadgetoid, 2020-01-23:
@dkinon possible road block- where do we obtain a set of previous Mopidy debs? I've done some frantic Googling and, as near as I can tell, the only source was their repository, and I can't seem to find a mirror or other source apart from - hypothetically- a local cache on a Pi perhaps.

Maybe using the debian supplied version is the sensible option, though this may present compatibility issues with more recent plugins?

Thanks again for your input on this, by the way, I feel stuck between a rock and a hard place with respect to our Mopidy-centric support of Pirate Audio boards.

kingosticks, 2020-01-23:
Nice, I'll try a give it all a go on mine tonight (with another webclient).

dkinon, 2020-01-23:
@Gadgetoid here is the list of mopidy repo packages I have on my working installation:
```
libspotify12-12.1.103-0mopidy1
mopidy-2.3.1-1
mopidy-spotify-3.1.0-0mopidy1
python-spotify-2.1.3-0mopidy1
```

I was able to fish those 4 debs out of /var/cache/apt/archives, let me know how you'd like me to deliver them.  Also @Gadgetoid @docmollo @jodal, let me know if I'm missing any packages.

Gadgetoid, 2020-01-23:
If you could bung them to me at phil@pimoroni.com that'd be ace! I'll get 'em hosted ASAP.

Gadgetoid, 2020-01-24:
Thanks to @dkinon we now have the 2.x packages hosted on get.pimoroni.com, which is the first step to putting together a stand-in installer that will keep us limping along until the whole stack is ready to move over to Mopidy 3.x (hopefully soon!)

The files are:

- https://get.pimoroni.com/mopidy/libspotify12_12.1.103-0mopidy1_armhf.deb
- https://get.pimoroni.com/mopidy/mopidy-spotify_3.1.0-0mopidy1_all.deb
- https://get.pimoroni.com/mopidy/mopidy_2.3.1-1_all.deb
- https://get.pimoroni.com/mopidy/python-spotify_2.1.3-0mopidy1_armhf.deb

```
sha256sum *.deb
52e69f7b08626a2b02474a2739cec59ad84498847c9b382e2e1d9e2d873eadd8  libspotify12_12.1.103-0mopidy1_armhf.deb
7baba9e19867e8d1df70e53620b838d8e43f34c62a548880c0882a746762dba2  mopidy-spotify_3.1.0-0mopidy1_all.deb
cb58771cd4a2427e4db62faee4c78a338cbfdfbc472a2b51134c13fa1cf3cfaa  mopidy_2.3.1-1_all.deb
416b596cf2c2aed283216766fe1c510b2ee8a458e0f81eeb4ea5aa64eb2c682e  python-spotify_2.1.3-0mopidy1_armhf.deb
```

The install script will have to wget these packages and use `dpkg` to install them in addition to- probably- installing `mopidy-iris==3.43.0`. (Operating on the assumption the next release might be Python3 only and/or have breaking changes.)

bradmason25, 2020-01-25:
@Gadgetoid I just tested the packages you hosted rather than the mopidy repo (with the latest iris which is still v3.43.0) and it's worked! Thank you very much

I had to resolve a slight problem with the GStream alsasink not being installed but that might be unrelated

kingosticks, 2020-01-25:
The gstreamer1.0-alsa
<https://packages.debian.org/it/sid/gstreamer1.0-alsa> package,
as well as -tools and -pulseaudio, are _recommended_ Mopidy dependencies
and are probably not considered when installing manually like this. You
need that -alsa package if you are specific the alsasink output.

kingosticks, 2020-01-26:
I'm up and running with the Mopidy 3 versions, everything seems to work great. The only install issue I hit was the following exception
```
Jan 26 13:30:29 raspberrypi mopidy[5265]:   File "/usr/local/lib/python3.7/dist-packages/PIL/Image.py", line 69, in <module>
Jan 26 13:30:29 raspberrypi mopidy[5265]:     from . import _imaging as core
Jan 26 13:30:29 raspberrypi mopidy[5265]: ImportError: libopenjp2.so.7: cannot open shared object file: No such file or directory
```
Which was solved by installing `libopenjp2-7` from APT. I don't understand why that dependency isn't handled by python-pil (or what the difference is between pil and pillow).

dkinon, 2020-01-27:
@kingosticks a good way to resolve dependency issues after installing debs via dpkg is to run `apt-get install -f` which will attempt to fix any broken dependencies.  This is the same approach I'm using in the updated installer script.

kingosticks, 2020-01-27:
This was a Mopidy 3 installation using apt, no dpkg here at all. I'm guessing the problem lies with using the older python3-pil Debian package as a way to fulfill the dependencies for newer releases from pip. But I'm just guessing.

dkinon, 2020-01-27:
@kingosticks ah, I see that now, apologies.  If you can open an issue on the finding with the mopidy project, I'm sure they would appreciate it.

kingosticks, 2020-01-27:
Sorry I wasn't that clear, python3-pil is a dependency for the mopidy-pidi extension, nothing to do with Mopidy itself.

techcom1, 2020-01-28:
First of all thank you very much for providing the above packages which solves my problems 
which mopidy running in buster!

But now i have issues with nginx. 
In my case i am running mopidy-party as guest-mode and mopidy-mobile as DJ-mode.
Access to mopidy-party with lan IP only (without port number).
Access to mopidy-mobile is restricted with login.
I realized this with nginx in conjunction with iptables and that worked over the last 2 years without
any troubles.

My config:
server {
        listen 80 default_server;
        server_name _;

        location /mobile {
		proxy_pass http://localhost:6680;
		proxy_http_version 1.1;
		proxy_set_header Upgrade $http_upgrade;
		proxy_set_header Connection "upgrade";
		auth_basic "Restricted";
		auth_basic_user_file /etc/nginx/.htpasswd;				
		}
				
        location / {
                proxy_buffering off;
                proxy_pass http://localhost:6680;
                proxy_http_version 1.1;
                proxy_set_header Upgrade $http_upgrade;
                proxy_set_header Connection "upgrade";
                rewrite /mopidy/(.*) /mopidy/$1 break;
                rewrite /(.*) /party/$1 break;
                proxy_redirect off;
		}
}

iptables drops port 6680.

works with below environment:
debian stretch
nginx: 1.10.3
mopidy 2.2.3
mopidy-spotify; 3.1.0
mopidy-party: 0.3.1
mopidy-mobile: 1.9.0

The issue now is, that the clients are loading the right pages, ask for credentials but at this stage
they are hanging (trying to connect but without success). 
The changelogs of mopidy said, that the version of torpedo have changed from 4.x to 5.x. 
Could this be the reason? 
Is it possible that mopidy has it's own redirect/forward function which interferes with nginx?
The question for me now is, where and how should i start examining the problem? 

Many Thanks in Advance.

kingosticks, 2020-01-28:
@techcom1 you'd be better off asking your question at https://discourse.mopidy.com/ rather than taking this issue off-topic. Plus it doesn't sound like your problem is anything to do with Pirate Audio.

techcom1, 2020-01-28:
@kingosticks 
Thx and sorry!

Gadgetoid, 2020-01-29:
It looks like Mopidy 3.x support for Iris is imminent - https://github.com/jaedb/Iris/issues/478#issuecomment-577065746

Gadgetoid, 2020-01-29:
@dkinon would appreciate your feedback on 7afb5c44a70e2bc6f4ed1f3f622531d568f18fb3

Have people (in house) jumping up and down about the Pirate Audio installer, so I had to steal your thunder and get a fix out post-haste.

kingosticks, 2020-01-29:
dpkg doesn't install any dependencies, that's up to you (or to use `apt install -f`. Where are you installing Gstreamer etc?

Gadgetoid, 2020-01-30:
> dpkg doesn't install any dependencies, that's up to you (or to use `apt install -f`). Where are you installing Gstreamer etc?

Groan. I'm not used t going to this kind of extent to side-load packages. I think this whole trick just became redundant though: https://github.com/jaedb/Iris/releases/tag/3.44.0

[4 more replies]

closed 2020-03-16 by Gadgetoid