pimoroni/pirate-audio #11: Anybody can attach to iris client

Rocketansky, 2020-01-06:
After the installation the small display shows the IP address with the port.
I accessed it with a web browser and got directed to the iris client.
Immediately I had access to the client and to files it controls.
As this is a WLAN connection anybody in the same WLAN could do that.

docmollo, 2020-01-06:
If you read the [docs for Mopidy](https://docs.mopidy.com/en/latest/), and look at your config file, you'll see that you need to edit the [allowed_origins config option](https://docs.mopidy.com/en/latest/ext/http/#confval-http/allowed_origins).

Rocketansky, 2020-01-07:
@docmollo I don't see how this solves the problem that the iris client is vulnerable as a puppy dog.

docmollo, 2020-01-07:
First, to be clear, Pimoroni has nothing to do with Iris. If you're not happy with Iris, [there are other options you could install](https://mopidy.com/ext/). Not sure if any of them have authentication options.

Looks like I mis-read what allowed_origins does. It has nothing to do with limiting the clients that can connect to the Mopidy web server.

If you're really concerned about who can connect to the web interface, you should look into setting up firewall rules.

Gadgetoid, 2020-01-24:
In the interest of clarity I'm labelling this as wontfix/documentation. Not so much because we don't maintain Iris, but because this configuration is intentional but I do think we might be able to better document why it's set up this way and what risks it might entail.

Security is a sensititve topic, but secure-by-default configurations are unfortunately not accessible, resulting in extra load on our support team (and, frankly, me).

I am, arguably to a fault, singularly focussed on making setup as easy and frictionless as possible for end users. Most people just want it to work quickly and simply.

I'm operating under the assumption that Iris is a very low risk security concern, and that those who prefer a secure configuration will know how to set it up.

Do you have any insights as to why you'd prefer a secure configuration, or what the risks (short of family members simply rickrolling each other) of this set up might be?

closed 2023-01-20 by Gadgetoid