pimoroni/pirate-audio #12: After installation web client shows "Mopidy-Spotify extension is not running"

Rocketansky, 2020-01-06:
After I got the installation correct I investigated the iris client.
Clicking on the settings for Spotify I saw an orange bar with "Mopidy-Spotify extension is not running  - you will not be able to play any Spotify tracks"

docmollo, 2020-01-06:
Sounds like you don't have the spotify plugin enabled. Check the spotify section of your config file. By default, the plugin is set to disabled (false).

Rocketansky, 2020-01-07:
@docmollo Thanks! I changed the value in file /etc/mopidy/mopidy.conf and it worked.

Should be part of some documentation - IMO

closed 2020-01-07 by Rocketansky

derekslenk, 2020-02-20:
I am also having this issue. However, I have enabled spotify in the mopidy.conf

Do we need to generate all the tokens for the plugin or is the web interface supposed to take care of that?