pimoroni/pirate-audio #13: No possibility to shutdown gracefully

Rocketansky, 2020-01-06:
Neither the buttons on the device itself nor the unfortunate web client offers a possibility to shutdown gracefully.
One has to connect to the raspberry installation to do so.

docmollo, 2020-01-06:
This issue would be solved by the long-term plans to create a read-only OS for this product.

Rocketansky, 2020-01-07:
@docmollo I don't see how a "read-only OS" would bring a power button on the hardware ;-)

docmollo, 2020-01-07:
Obviously, it would not. It would allow you to remove power without having to worry about corrupting the OS.

Gadgetoid, 2020-01-24:
It may be worth asking over at the Iris GitHub for a "Shutdown" button. It's not the silliest idea.

There's some related- albeit probably unhelpful- discussion here: https://github.com/jaedb/Iris/issues/382

We kept power control functionality out of Pirate Audio boards since it wasn't worth pushing the price up for a feature only a subset of people will use and we already make a "SHIM" to solve this very problem: https://shop.pimoroni.com/products/onoff-shim

The bonus with the SHIM is that you can turn your Pi back on without having to disconnect and reconnect the power.

duracell80, 2021-01-28:
I'm working on a menu based UI that will include safe shutdown and reboot.

closed 2023-01-20 by Gadgetoid