pimoroni/pirate-audio #15: pirate-audio display for Kodi

zicmama, 2020-01-13:
Any ideas to help using pirate-audio display with Kodi.
https://www.opendisplaycase.com/kodidisplayinfo-program.html

Gadgetoid, 2020-01-24:
I guess if Kodi Display Info still works it at least shows it's possible. Unfortunately I don't have the bandwidth to support any additional platforms. Keeping up with Mopidy is already hard enough!

Displaying X media info on Y display is the domain of PiDi ("Pirate Display" - https://github.com/pimoroni/pidi). Or at least intended to be. PiDi is *very* strongly minded toward music but I don't see any reason why a client plugin couldn't make it work for Kodi.

You should raise an issue over at https://github.com/pimoroni/pidi/issues and hope someone swoops in (it might be me, when I'm less busy) and adds support.

hgroll, 2020-04-28:
Maybe you have more success with KodiDisplayInfo if you use the PiDi as a frame buffer device (driver/staging/fbtft/fb_st7789v.ko) under Linux. I had a bit of success (with some flaws) with this [device tree overlay](https://gist.github.com/hgroll/2731ae6d05350df663b123615f765bf5) I've put together.

Just compile it with the device tree compiler as an overlay:
`dtc -@ -I dts -O dtb -o pidi.dtbo pidi-overlay.dts`
place it in _/boot/overlays/pidi.dtbo_ and include the overlay in the raspberry pi config _/boot/config.txt_:
`dtoverlay=pidi`
Tested with _Linux pi 4.19.66-v7+_ _Raspberry Pi 2 Model B Rev 1.1_.