pimoroni/pirate-audio #16: Add BCM24 to button config to support newer Pirate Audio boards

sandyjmacdonald, 2020-01-24:
On new versions of the Pirate Audio boards (after 23 January 2020), the button previously connected to BCM20 is now connected to BCM24. Therefore, the line `bcm24 = volume_up,active_low,250` needs to be added to `install.sh` to add support for this button.

PaulWebster, 2020-01-26:
Can someone who has one of these boards - old or new - tell the difference through inspection?

Gadgetoid, 2020-03-10:
@PaulWebster not easily- probably the simplest way is to set up a Python script that reads the two GPIO inputs, press the button and see which fires. `buttons.py` should do the trick - https://github.com/pimoroni/pirate-audio/blob/master/examples/buttons.py

closed 2020-03-10 by Gadgetoid