pimoroni/pirate-audio #17: Add support for Spotify Connect

moritz1000, 2020-01-26:
I bought a Pirate Audio device because I thought it supports Spotify Connect but it turns out I can stream Spotify to it but no via Spotify Connect.

I could not find a way to add Spotify Connect to Mopidy.

I tried achieving this with Volumio since it supports external displays and GPIO control via plugins and also Spotify Connect but I could not even get the sound output to work.

To summarize I think Spotify Connect would be a great feature but I am unable to get it working.

Gadgetoid, 2020-01-30:
I g uess both Volumio and Mopidy being music player daemons basically means they hog the audio ouput device and don't give things like raspotify a look in.

You should probably just install raspotify onto a fresh Raspbian and see how you go with that: https://github.com/dtcooper/raspotify

All of the display code we've got working thus far is Mopidy/mpd-centric and wont work with Raspotify. There surely *must* be a way to query Librespot for the currently playing song/album art/etc though.

Gadgetoid, 2020-01-30:
Okay it looks like librespot (the grunt behind raspotify) supports an `--onevent` option that lets you configure it with an endpoint for handling track start/change/stop events: https://github.com/librespot-org/librespot/issues/185

The event type and releated track IDs are saved to environment variables:

https://github.com/librespot-org/librespot/blob/aa880f8888226a8e5fc6e1e54dfb7cf58176ac95/src/player_event_handler.rs#L18-L33

So all that remains - okay so this isn't exactly trivial - is to write a small Python handler that:

1. Reads these environment variables
2. Looks up the TRACK_ID on Spotify's API service
3. Retrieves the track name, album art, etc
4. Displays it on the Pirate Audio display
5. Exits, leaving the display unchanged until the next event

Gadgetoid, 2020-01-30:
This all said. VolumIO should work- I guess you set up the DAC in the System settings? I think the missing link is the enable pin. If you can edit `/boot/config.txt` then adding the line `gpio=25=op,dh` should get you up and running.

moritz1000, 2020-01-30:
Thank you for talking your time and writing such a detailed response. 

>  I guess you set up the DAC in the System settings?

I looked for something called "dtoverlay=**hifiberry-dac**" as used for pirate audio but did not add `gpio=25=op,dh` to `/boot/config.txt`. There was a list of many different hifiberry dacs in VolumIO and I tried some of them.

First of all I will try to get Rasbian + raspotify + sound output working and then look into the rest but I guess this goes way over my head.

dogchunx, 2020-01-30:
I've been curious about this too and just had a quick go at getting Spotify Connect playing through the Pirate Audo DAC via Raspotify. 

Turns out all I needed to do run the "Easy" installation instructions as detailed in the Raspotify project page and then change the options in the config file to use the correct audio device, which in my case was:

`OPTIONS="--device hw:1,0"`

The LCD and buttons don't function doing it this way, but to anyone who just wants a basic way of controlling what Pirate Audio is playing via Spotify Connect, it doesn't seem like a bad solution. This also doesn't touch Mopidy, but this might be moot as what's playing can now be controlled from the Spotify client on any desktop or mobile device.

moritz1000, 2020-01-31:
Did you add `OPTIONS="--device hw:1,0"` in `/etc/default/raspotify`? Thank you for testing this.

dogchunx, 2020-01-31:
Hey, no probs. That's where I added it, yep.

shokwaav, 2020-05-07:
Does raspotify show the media info on the LCD?

moritz1000, 2020-05-07:
> 
> 
> Does raspotify show the media info on the LCD?

Sadly no. I still think that a real Spotify Connect integration would be so useful.

Abysmax, 2020-06-20:
+1 I actually think this would be a great feature.

Ennneerrrgggyyy, 2020-08-23:
Wow - I am really grateful and impressed how quickly you can hack something like this. I've already learned a lot today. I hope that it is okay, to ask so many questions.

Is started with adding 
```
dtoverlay=hifiberry-dac
gpio=25=op,dh 
```
to /boot/config.txt. Afterwards I executed the following:
```
sudo pip3 install spotipy pidi_display_st7789
mkdir -p /home/pi/spotipy/cache/
touch /home/pi/spotipy/hook.log
sudo chown raspotify:raspotify /home/pi/spotipy/hook.log
sudo usermod -a -G gpio raspotify
sudo usermod -a -G spi raspotify
```
Then I added
```
OPTIONS="--device hw:1,0 --onevent /home/pi/spotipy/hook.sh"
```

and executed
```
sudo systemctl restart raspotify
```

Afterwards I executed:
```git clone https://github.com/pimoroni/pidi-spotify
cd pidi-spotify
python3 -m pidi_spotify
```

Somehow the Terminal gets unresponsive at this point
```
pi@raspberrypi:~/pidi-spotify $ python3 -m pidi_spotify
open 5
```

Also the problem persists, that I can't connect through Spotify Connect when "--onevent /home/pi/spotipy/hook.sh" is added. When I add only OPTIONS="--device hw:1,0" I can connect properly.

Gadgetoid, 2020-08-23:
You're most of the way there! I should mention that to test `pidi_spotify` you will need a second terminal Window, because the running application will block that terminal. You can exit with Ctrl+C. I haven't turned it into a system-service-like-thingy just yet.

I encountered the problem with adding a hook, where I wasn't able to log in. It could be that Raspotify cannot run `hook.sh`

Try: `chmod +x /home/pi/spotipy/hook.sh`

And make sure your `hook.sh` is updated ot the new one included with `pidi-spotify`.

Ennneerrrgggyyy, 2020-08-23:
Shouldn't the new script generate the new hook.sh? Or do I still have to sudo nano generate it?

Is it correct that "python3 -m pidi_spotify" kind of freezes with open 5?

```
pi@raspberrypi:~/spotipy $ chmod +x /home/pi/spotipy/hook.sh
chmod: changing permissions of '/home/pi/spotipy/hook.sh': Operation not permitted
```

Gadgetoid, 2020-08-23:
You'll have to copy the "hook.sh" from the `pidi-spotify` directory copied down with git.

So assumping you're somewhere like: `/home/pi/pidi-spotify` you can `cp hook.sh /home/pi/spotipy/hook.sh`

And try: `sudo chmod +x /home/pi/spotipy/hook.sh` ... the Pi likes this simon-says nonsense 😆

Ennneerrrgggyyy, 2020-08-23:
oh boy! that did it. and it works really good. 😍

Ennneerrrgggyyy, 2020-08-23:
So at the moment the whole thing only works while "python3 -m pidi_spotify" is running, right?

This is what you meant when you wrote that you haven't turned it into a system-service-like-thingy just yet, right?

Gadgetoid, 2020-08-23:
That's correct! Yes. I need to create a way to configure pidi_spotify and write a script to run it as a service.

You can probably create a script called `/home/pi/spotipy/start.sh` and put something like this in it (changing values and/or directories as needed):

```bash
#!/bin/bash

cd /home/pi/pidi_spotify
export SPOTIPY_CLIENT_ID="YOUR CLIENT ID"
export SPOTIPY_CLIENT_SECRET="YOUR CLIENT SECRET"
python3 -m pidi_spotify
```

Then add it to your crontab by typing:

```
crontab -e
```

And then adding:

```
@reboot /home/pi/spotipy/start.sh
```

When it's finished I hope `pidi_spotify` will just be a program you can run *either* as a system service, or as a utility to control that system service, so we can get rid of `hook.sh` altogether and just have a command like `pidi_spotify --raspotify` do everything that hook does.... in theory anyway! This is all an elaborate proof of concept.

Gadgetoid, 2020-08-23:
Right I've added instructions to the PiDi Spotify repository. Any other questions/issues/suggestions/requests then please raise them in a new issue over there: https://github.com/pimoroni/pidi-spotify

For those wanting to use Raspotify/Spotify Connect with Mopidy then you will still need to use my first soution above which I will re-iterate here:

```
sudo apt install mopidy-mpd mpc
```

Then use the below script as your `--onevent` hook for Raspotify, by editing `/etc/default/raspotify` and making your options line look something like this:

```
OPTIONS="--device hw:1,0 --onevent /home/pi/raspotify.sh"
```

Save this as `/home/pi/raspotify.sh` and ensure it's executable with `sudo chmod +x /home/pi/raspotify.sh`:

```bash
#!/bin/bash

if [ "$PLAYER_EVENT" = "start" ]; then
        mpc clear
        mpc add spotify:track:$TRACK_ID
        mpc play
        sleep 1
        mpc pause
elif [ "$PLAYER_EVENT" = "change" ]; then
        mpc clear
        mpc add spotify:track:$TRACK_ID
        mpc play
        sleep 1
        mpc pause
elif [ "$PLAYER_EVENT" = "stop" ]; then
        mpc play
elif [ "$PLAYER_EVENT" = "volume_set" ]; then
        mpc_volume=$(($VOLUME * 100 / 65536))
        mpc volume $mpc_volume
fi
```

robrecord, 2020-09-10:
pimoroni/pidi-spotify is all I need for my pirate-audio-enabled Pi 2; that and shairport-sync. Thank you Gadgetoid!