pimoroni/pirate-audio #19: Fixes for mopidy install script

sandyjmacdonald, 2020-02-04:
1. Pending a new release of Iris on PyPI, do the following in the install script:

`python3 -m pip install https://github.com/jaedb/Iris/archive/develop.zip`

2. Add support in the config file for the changed button pin in newer revisions of the Pirate Audio boards (more info in #16 ):

`bcm24 = volume_up,active_low,250`

3. Remove lines with semicolon comments in the config file, as they seem to bork it in more recent Mopidy versions.

sandyjmacdonald, 2020-02-05:
Until a release of Iris on PyPI, you can give this modified install script a go, which does the above. It works for me on the latest version of Raspbian Buster (2019-09-26).

https://gist.github.com/sandyjmacdonald/b72b65d827d85166b4739e12867d9e1b

jplegat, 2020-02-07:
It did work on a clean Buster installation.

However, if you follow the instructions on [https://learn.pimoroni.com/tutorial/sandyj/getting-started-with-pirate-audio](https://learn.pimoroni.com/tutorial/sandyj/getting-started-with-pirate-audio) to add local files it doesn't work because the Mopidy-Local extension was not installed in the process. 

`sudo python3 -m pip install Mopidy-Local` did the trick.

Thanks for updating the script!

GregJohnStewart, 2020-02-11:
Sounds like that should be part of the script then? Worked mostly, except the default http config is broken. (doesn't seem to like a blank hostname)

blomsoft, 2020-02-11:
I use the IP address i.e. 192.168.1.42 in my case.

Gadgetoid, 2020-02-11:
I could swear that `0.0.0.0` worked for me. It certainly did in Mopidy 2.x so this is probably a Mopidy bug. I have a vague recollection of this being discussed elsewhere, but my git-search-fu is not serving me well today.

GregJohnStewart, 2020-02-13:
Figured it out, it did like `0.0.0.0`, but I noticed this issue: https://github.com/mopidy/mopidy/issues/1868

After removing the inline client it worked just fine.

Gadgetoid, 2020-02-13:
Aha, that's the one- thank you! In all this chaos I'd forgotten to remove the inline comments that Mopidy 3.x broke.

blomsoft, 2020-02-13:
Thanks, confirmed to work with 0.0.0.0 without inline trailing comment.

kristerj, 2020-02-22:
This is super helpful. To sum up, to get running on a fresh buster install with the steps in the tutorial that expect the local module:

1. Install python3
2. Modify install script
  -install from develop.zip
  -remove semicolons 
  -pin change 
3. Pip install local module

Is that correct?

Gadgetoid, 2020-02-23:
You should be able to combine 2. into just "run the install script" since Mopidy Iris is now released onto PyPi and i've fixed the config.

closed 2020-03-09 by Gadgetoid