pimoroni/pirate-audio #23: Line out card - customise volume levels

tennisparty, 2020-04-24:
Is there a way of altering the default volume scale and maximum volume? The board outputs at such a high volume it is only useable on several of my amps when at around 10% volume. 

The audio section of my mopidy conf file is as follows
[audio]
mixer_volume = 20
output = alsasink device=hw:sndrpihifiberry

I tried adding a alsamixer section but it has made no difference
[alsamixer]
card = 1
control = PCM
min_volume = 0
max+voume =20
volume_scale = cubix

jjok, 2020-04-26:
Maybe related to mopidy/mopidy#1870?

chee, 2020-12-01:
i added a volume control to mine by setting up softvol in `/etc/asound.conf`:

```conf
pcm.monkey { 
  type hw card 0 
}

pcm.dmixer { 
  type dmix 
  ipc_key 1024 
  slave { 
    pcm "monkey" 
    channels 2 
  } 
}

ctl.dmixer { 
  type hw 
  card 0 
}

pcm.!default {
    type            plug
    slave.pcm       "softvol"
}

pcm.softvol {
    type            softvol
    slave {
        pcm         "dmixer" 
    }
    control {
        name        "PCM"    
        card        0
    }
}
```

this setup also includes dmix, so multiple processes can use the soundcard at once

you can read more about softvol here: https://alsa.opensrc.org/Softvol

closed 2023-01-20 by Gadgetoid