pimoroni/pirate-audio #24: Radio Text not shown in Display

ingelumer, 2020-05-06:
When running a Radio Stream in Mopidy I can see the Radio Text in my web browser GUI like Title and Interpret (actually also in the Mopidy Mobile Android App), but the PirateAudio Display does only show the Radio Station name.
Would it be possible to show additional Information for Radio Steams just like for playing MP3/Spotify?

ingelumer, 2020-05-06:
just realized that when mentioning the WebGUI, Iris is not showing the Information as well, but musicbox_webclient does.

Gadgetoid, 2020-05-15:
Try `sudo pip3 install --upgrade mopidy-pidi` and see if the new release works for you.

ingelumer, 2020-05-15:
Looks indeed much better, thanks for the hint!
I can see the Artist and Title now, but the Radio Station Name is gone :-(

closed 2023-01-20 by Gadgetoid