pimoroni/pirate-audio #26: extending use of Spotify

Nico-L, 2020-05-10:
Hello,

First, thank you for these boards and the effort put in the software. I bought two of the 3W dac boards to make two "Spotify box" for my kids. The idea is for them to access Spotify without their phone or tab screen.

I'd like to make a box around the boards and I am planning to add all the needed buttons to make the use as smooth as possible. And also, if needed, to try to code anything I'd need. But, if I have some knowledge in javascript, I am quite the newbie regarding Python.

So first thing first, I installed your software around mopidy which is a great start. The Spotify part is not easy to navigate through though and I'd need to add some new functionnalities, for example to directly access Spotify playlists.

But before digging into the code, I looked to other systems. I found piCorePlayer and [this thread](https://forums.slimdevices.com/showthread.php?111502-Jivelite-on-a-Pirate-Audio-240x240-screen&p=971970&viewfull=1#post971970) to install it for pirate audio boards. It is not completely trivial to setup but doable and I succeeded to have everything working. This player takes advantage of Tiny Core Linux a lightweight linux distribution and add audio around squeezelite system. But to use Spotify it also needs the Logitech Media Server, which makes the booting time skyrocketing. Also, the screen output is too complicated for the tiny screen of you board. Full navigation in menus and submenus are mandatory to get to Spotify. I have a working solution, albeit too complicated for my taste.

I am willing to have a try at coding a bit around your solution. But as I said, I have next to no knowledge about Python and this may be way over my skills and I have some very basic questions to kickstart.
* what environment I'd need to code properly? what is the process of local testing modification on the hardware?
* I read the code and came to the conclusion that adding buttons would require to modify mopidy-raspberry-gpio package and add new functions in the frondend.py (if I can find the right mopidy command). Is there something I am missing?
* the volume buttons have a delay of about 1s before the commands take effect, is there something that can be done? Is it on the mopidy side?

Thanks

Gadgetoid, 2020-05-12:
Mopidy-PiDi and Mopidy-Raspberry-GPIO are both standalone plugins that don't directly interact, so to build a graphical interface with playlist browsing you would effectively be chucking these away and starting from scratch. For this you'd have to build a "frontend" extension to Mopidy, and brave the Mopidy extension development docs: https://docs.mopidy.com/en/latest/extensiondev/#extensiondev

Looking at Mopidy Touchscreen might help - https://github.com/9and3r/mopidy-touchscreen

Honestly I don't think the Pirate Audio boards have enough screen resolution and inputs to give you anything approaching a usable Spotify interface.

The volume delay is built into Mopidy- it transitions the volume to the new level when a change is requested.

As for development environment, you'd probably want to start with a self-contained prototype using PIL to try and render and control a GUI to see if it's feasible. I'm not well versed in Mopidy, or how the Iris front-end gets the information that it does from Spotify so I can't be of more specific help.

Nico-L, 2020-05-12:
Hi,
Thank you for taking some time to answer me.

Actually, the piCorePlayer showed that you're right: the screen is not big enough to propose a smooth navigation. I have a simpler way of handling Spotify in mind : adding, let's say 4 buttons linked to the 4 first Spotify playlist if that's possible, or add 4 playlist IDs in the conf file. Each button would directly play the first track of the corresponding playlist. For this, it seems that I just have to add some functionnalities (and find the right commands from mopidy) to Mopidy-Raspberry-GPIO, doesn't it?

The screen would still be used for info display (though at some point I might want to let more place to images and less to info, but this is not critical for now).

I'll look in the extension doc to start to tinker while waiting for my protozero to arrive.

Thanks

Nico-L, 2020-05-13:
Hi,

I found this [thread](https://forums.pimoroni.com/t/pirate-audio-extending-raspberry-gpio-mopidy-plugin/13042/6) that basically proposes the same thing. I posted over there to have some more info. 

Closing hereas there is no need to keep this open. Thanks @Gadgetoid for your answers.

closed 2020-05-13 by Nico-L