pimoroni/pirate-audio #27: Pirate Radio SMOL Speaker – Doesn't work with RPi 4

KevinVitale, 2020-06-09:
Hi team. The Pirate Radio SMOL Speaker refuses to play audio on RPi 4.

### What doesn't work
 - The speaker will not play audio on my RPi 4, 4GB rev. 1.2;
 - I've tested the HAT on **two different boards**, both RPi 4, 4GB rev. 1.2;
 - I've reinstalled RPi OS at least 3 times, following the instructions precisely.
 - RPi OS: _Raspberry Pi OS (32-bit) Lite; SHA-256: f5786604; Release Date: 2020-05-27_

I've performed all necessary steps, incl. `dtparam=audio=off`.

```
➜  ~ sudo lshw -short
H/W path      Device  Class      Description
============================================
                      system     Raspberry Pi 4 Model B Rev 1.2
/0                    bus        Motherboard
/0/1                  processor  cpu
/0/2                  processor  cpu
/0/3                  processor  cpu
/0/4                  processor  cpu
/0/5                  memory     3906MiB System memory
/0/0                  bridge     Broadcom Limited
/0/0/0                bus        VL805 USB 3.0 Host Controller
/0/0/0/0      usb1    bus        xHCI Host Controller
/0/0/0/0/1            bus        USB2.0 Hub
/0/0/0/0/1/3          input      Dell USB Wired Multimedia Keyboard
/0/0/0/1      usb2    bus        xHCI Host Controller
/1            eth0    network    Ethernet interface
/2            wlan0   network    Wireless interface
```
<br/>

### What does work
 - RPi Zero WH
 - Using the same SD card
 - Using the same Pirate Audio SMOL Speaker HAT

```
➜  ~ sudo lshw -short
H/W path  Device  Class      Description
========================================
                  system     Raspberry Pi Zero W Rev 1.1
/0                bus        Motherboard
/0/0              processor  CPU
/0/1              memory     432MiB System memory
/1        usb1    bus        DWC OTG Controller
/1/1              generic    AX88179
/2        eth0    network    Ethernet interface
/3        wlan0   network    Wireless interface
```

<br/>

**PLEASE NOTE**: I also own the Pirate Audio Headphone HAT, and when I place this on my RPi 4, audio works as expected, albeit through the headphone jack.

Please advise

Gadgetoid, 2020-06-09:
On both devices what do you see if you run `aplay -l`?

For testing the sound you should use `speaker-test -twav -c2` to eliminate any interim software from the equation.

I've imaged a new Pi OS SD card for testing and confirmed audio out is working with a Pi Zero W, but switching over to my Pi 4 invokes some kind of WiFi-No-Worky error that I'm going to have to tackle tomorrow.

I also need to get my hands on the various Pirate Audio boards or hand this over to tech support, since I only have the Speakers version at home.

Both boards use the same exact i2s audio DToverlay and, IIRC, the same amp enable pin so it's decidedly baffling that one doesn't work on Pi 4 while the other does. I've seeen weirder problems, though!

ondrejtoral, 2020-06-14:
I've been dealing with the same issue on RPi 3B+ and I have found out why it did not play for me - I was using the Fan SHIM, which blocks the port enabling DAC converter. (It is clearly written on the Fan SHIM product page). Uninstalling the Fan SHIM library enabled audio playback.

KevinVitale, 2020-06-15:
@Gadgetoid Testing `speaker-test -twav -c2` produces no sound on RPi.

FWIW, I placed the HAT on a RPi 3B+, inserted the exact same microSD card as before, booted up and ran `speaker-test -twav -c2`, which produced sound. So both the Zero WH & RPi 3B+ work; RPi 4B doesn't...all with the same microSD card.

closed 2023-01-20 by Gadgetoid