pimoroni/pirate-audio #28: pirate-audio/mopidy/README.md  Config File disables mpd & http

ed-dd, 2020-07-29:
Using the given /etc/mopidy/mopidy.comf gives a number of errors including not being able to use a browser to connect to iris frontend unless using the full url.  Also, Iris frontend System>Advanced>Connection  'Not Connected' error msg.

-- Logs begin at Wed 2020-07-29 16:44:08 AEST, end at Wed 2020-07-29 17:17:01 AEST. --
Jul 29 16:44:36 raspberrypi systemd[1]: Starting Mopidy music server...
Jul 29 16:44:36 raspberrypi systemd[1]: Started Mopidy music server.
Jul 29 16:45:41 raspberrypi mopidy[349]: INFO     [MainThread] mopidy.__main__ Starting Mopidy 3.0.2
Jul 29 16:45:41 raspberrypi mopidy[349]: INFO     [MainThread] mopidy.config Loading config from builtin defaults
Jul 29 16:45:41 raspberrypi mopidy[349]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d/mopidy.conf
Jul 29 16:45:41 raspberrypi mopidy[349]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d/mopidy-local.conf
Jul 29 16:45:41 raspberrypi mopidy[349]: INFO     [MainThread] mopidy.config Loading config from file:///etc/mopidy/mopidy.conf
Jul 29 16:45:41 raspberrypi mopidy[349]: INFO     [MainThread] mopidy.config Loading config from command line options
Jul 29 16:45:41 raspberrypi mopidy[349]: WARNING  [MainThread] mopidy.config Ignoring config section 'mpd' because no matching extension was found
Jul 29 16:45:50 raspberrypi mopidy[349]: INFO     [MainThread] mopidy.__main__ Enabled extensions: local, iris, file, softwaremixer, pidi, stream, raspberry-gpio, m3u
Jul 29 16:45:50 raspberrypi mopidy[349]: INFO     [MainThread] mopidy.__main__ Disabled extensions: http, spotify
Jul 29 16:45:50 raspberrypi mopidy[349]: WARNING  [MainThread] mopidy.__main__ Found http configuration errors. The extension has been automatically disabled:
Jul 29 16:45:50 raspberrypi mopidy[349]: WARNING  [MainThread] mopidy.__main__   http/hostname must be a resolveable hostname or valid IP
Jul 29 16:45:50 raspberrypi mopidy[349]: WARNING  [MainThread] mopidy.__main__ Please fix the extension configuration errors or disable the extensions to silence these messages.
Jul 29 16:45:54 raspberrypi mopidy[349]: INFO     [MainThread] mopidy.commands Starting Mopidy mixer: SoftwareMixer
Jul 29 16:45:55 raspberrypi mopidy[349]: INFO     [MainThread] mopidy.commands Mixer volume set to 40
Jul 29 16:45:55 raspberrypi mopidy[349]: INFO     [MainThread] mopidy.commands Starting Mopidy audio
Jul 29 16:45:55 raspberrypi mopidy[349]: INFO     [MainThread] mopidy.commands Starting Mopidy backends: FileBackend, M3UBackend, StreamBackend, LocalBackend
Jul 29 16:45:55 raspberrypi mopidy[349]: INFO     [Audio-2] mopidy.audio.actor Audio output set to "alsasink device=hw:sndrpihifiberry"
Jul 29 16:45:55 raspberrypi mopidy[349]: INFO     [MainThread] mopidy.commands Starting Mopidy core
Jul 29 16:45:55 raspberrypi mopidy[349]: INFO     [MainThread] mopidy.commands Starting Mopidy frontends: RaspberryGPIOFrontend, PiDiFrontend, IrisFrontend
Jul 29 16:45:56 raspberrypi mopidy[349]: INFO     [IrisFrontend-10] mopidy_iris.core Starting Iris 3.50.0

Removing the comment and syntax from [mpd] & [http] fixed this.

[BEFORE]
[mpd]
hostname = 0.0.0.0 ; Allow access to mpd from other devices
[http]
hostname = 0.0.0.0 ; Allow access to HTTP/Iris from other devices

[AFTER]
[mpd]
hostname = 0.0.0.0 
[http]
hostname = 0.0.0.0

Gadgetoid, 2020-07-29:
How did you fetch/run this installer or find this config? Afaik we haven't had comments in the standard config for about 6 months.

They were removed here: https://github.com/pimoroni/pirate-audio/commit/cda077e7fbed7cd1272b0b5fdb3f6fdd1c8510db

Here's the bit that injects the config:
https://github.com/pimoroni/pirate-audio/blob/6915d4e61d6f8960f465d4b954e260342645424c/mopidy/install.sh#L172-L176

Gadgetoid, 2020-07-29:
Oh- I think I've found it! Did you grab this config from the README? Looks like that has not been updated.

closed 2020-08-19 by Gadgetoid