pimoroni/pirate-audio #30: Pinouts.

jx48z, 2020-08-20:
Hi there everyone, 
Newbie here

Wasn't entirely sure where to ask this question as its not specifically to do with the code.

I am currently converting an old cassette player to use the DAC line out on my pi zero w. 
I want to have the line out pointing out the case but don't have enough space to 'sandwich' the pi and the hat together. I am also using 4 basic other GPIO's for buttons.

Therefore, i need to connect the pi and the hat via cables.

I do not need to use the screen or buttons on the hat.

I have looked for pinouts but am not sure which ones need to be connected for audio exclusively, if anyone is familiar with the pinouts please let me know which ones need to be connected to get audio and then potentially what code i need to use to make it work with audio only.

Thank you so much, i know this question will probably get closed as its not an issue but if you could point me in the right direction of resources that would be great.

Gadgetoid, 2020-08-20:
The pins marked i2s on this pinout are the ones you'll need to connect- https://pinout.xyz/pinout/pirate_audio_headphone_amp#

You will also need +5v power and enough ground pins. The pinout suggests physical pin 25 and 39 but I can't recall if this is correct.

jx48z, 2020-08-20:
And will the pi recognise it as an audio output automatically? Or is there a program i need to run?

Gadgetoid, 2020-08-20:
You need to make sure you add `dtoverlay=hifiberry-dac` to `/boot/config.txt` since there's no auto-detection for i2s devices. It just throws the data into oblivion and hope's something is listening!

jx48z, 2020-08-20:
Thank you so much for your help.

jx48z, 2020-08-20:
Sorry one last thing, do i need to have the amp on pin connected?

Gadgetoid, 2020-08-20:
Oh, yikes, yes you will either need to connect it to the Pi or pull it high yourself with a resistor to +3.3v.

If you connect it to the Pi you will need to additionally add `gpio=25=op,dh` to your `/boot/config.txt` to ensure that pin is pulled high to switch on the amp.

closed 2023-01-20 by Gadgetoid