pimoroni/pirate-audio #32: Volume control without Mopidy

tomaculum, 2020-09-06:
I did set up my Pi Zero with the Pirate Audio Line In HAT and installed it as stated in the README but I have skipped the Mopidy step since I only want to use Raspotify.

Volume Control inside Spotify Connect does work but the Pirate Audio HAT outputs the volume at 100% and my sound system needs to be carefully set to its lowest possible volume and the Spotify Volume needs to be below 50%.

Did anyone got any kind of volume control to work without Mopidy?

Gadgetoid, 2020-09-07:
I think you can set the system volume with `alsamixer`, though I can't recall if this setting survives a reboot.

It should be relatively trivial to create a crude Python script that binds some buttons to vol up/down via alsamixer.

tomaculum, 2020-09-20:
Thank you for your reply!

If I open alsamixer it says: 

> This sound device does not have any controls.

The selected card is `snd_rpi_hifiberry_dac`

tomaculum, 2020-09-21:
I did solve my issue with the "softvol" option in alsa and I added an additional alsa equalizer with the help of those resources:

https://www.hifiberry.com/docs/software/guide-adding-equalization-using-alsaeq/
https://www.raspberrypi.org/forums/viewtopic.php?t=213459
https://learn.adafruit.com/adafruit-speaker-bonnet-for-raspberry-pi?view=all#add-software-volume-control-2711421-10

My working `/etc/asound.conf`:

```
pcm.hifiberry {
   type hw card 0
}

pcm.dmixer {
   type dmix
   ipc_key 1024
   ipc_perm 0666
   slave {
     pcm "hifiberry"
     period_time 0
     period_size 1024
     buffer_size 8192
     rate 44100
     channels 2
   }
}

ctl.dmixer {
    type hw card 0
}

pcm.softvol {
    type softvol
    slave.pcm "dmixer"
    control.name "PCM"
    control.card 0
}

ctl.softvol {
    type hw card 0
}

pcm.!default {
    type             plug
    slave.pcm       "equalizer"
}

ctl.!default {
  type hw
  card 0
}

pcm.equalizer {
  type equal
  #slave.pcm "plug:dmixer"
  slave.pcm  "plug:softvol"
}

ctl.equal {
  type equal;
}
```

To make it work with Raspotify, I've followed those steps:

https://github.com/dtcooper/raspotify/issues/23#issuecomment-476951906

And for shairport-sync:

https://github.com/mikebrady/shairport-sync/issues/878#issuecomment-617649514

closed 2020-09-21 by tomaculum