pimoroni/pirate-audio #37: ValueError No input device matching mic_out

PabloLeon, 2020-12-03:
When running cliprecord.py after installation (./install.sh) on the pirate audio dual-mic I get the above error. All other examples work well. Any advice?

66point99, 2021-02-14:
Me too - same "mic_out" error from 'cliprecord.py' + other (screen test) examples work.

Noticed that './install.sh' creates the file '/root/.asoundrc', which differs from the README.

Once I copied that file to '~/.asound.conf' and '/etc/asound.conf', and rebooted, running 'cliprecord.py' now gives error:
"ValueError: No output device matching 'upmix'" (hey, at least it's a different error!)

README suggests that 'upmix' is provided by pulseaudio. I have the latest pulseaudio installed.

Now what?

(P.S. I'm using a ZeroW, in case that's material)

PabloLeon, 2021-03-31:
I've looked a bit into this again and still to no avail. If I run cliprecord.py with sudo I get the upmix error, without sudo the mic_out error. Using the sounddevice python library directly seems to work fine. So I assume that some of the pi config isn't set up correctly? Any suggestions would be greatly appreciated!

Gadgetoid, 2021-03-31:
Linux audio is a garbage fire and Raspberry Pi OS updated recently-ish to default to Pulse Audio, so this has no doubt rained on my parade (it took me far too long to get this thing working in the first place, sigh). I'll look into this when I can, but it's unlikely to be *soon*... I have *quite* a backlog.

PabloLeon, 2021-04-01:
Thanks! Any advice would be a greatly appreciated!

megolla, 2021-04-12:
I just installed an old Version ("raspbian-stretch-lite") on my zero hoping that this would help – unfortunately without success! 

I still get errors when using: python3 cliprecord.py. Even the test "arecord -Dmic_out -c2 -r48000 -fS32_LE -twav -d5 -R10000 -Vstereo test.wav" throws the error "ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM mic_out"…

**Pimoroni Development Team:** Please update your installer – I think it is not fair to sell the "Pirate Audio: Dual Mic" and not supporting it properly!

PelzKo, 2021-05-15:
One thing why my raspberry was not finding the "mic_out" device was because I added the "dtoverlay=adau7002-simple" line before "dtoverlay=hifiberry-dac" and "gpio=25=op,dh" in the /boot/config.txt

Will not work:
dtoverlay=adau7002-simple
dtoverlay=hifiberry-dac
gpio=25=op,dh

Will work:
dtoverlay=hifiberry-dac
gpio=25=op,dh
dtoverlay=adau7002-simple

So maybe think about adding hifiberry-dac and op,dh to the clip recorder install script so that it is added automatically in the correct order. Because on the store page (https://shop.pimoroni.com/products/pirate-audio-dual-mic) it says to just clone the repository and run the install script and than it will work which is not the case.

MclarenFM, 2021-06-07:
Hi,
I just bought the dual mic hat and I exactly have this problem with Raspberry Pi Zero W!!!
I received this error on "cliprecord.py":
"ValueError: No output device matching 'upmix'"
No one could find a solution for this issue yet???
I wish I read this topic before buying the product :-( !
Kind Regards.

Gadgetoid, 2021-06-15:
@MclarenFM 

> I just bought the dual mic hat and I exactly have this problem with Raspberry Pi Zero W!!!

Which OS?

Does @PelzKo's fix work?

> **Pimoroni Development Team:** Please update your installer – I think it is not fair to sell the "Pirate Audio: Dual Mic" and not supporting it properly!

@megolla up until recently that "team" was just me :grimacing: I don't like to point that out because it's our problem not yours, but I kinda owe some explanation for why this stuff takes *sooooooooooooo* long for me to get 'round to and - for better or worse - that's the size of it.

Gadgetoid, 2021-06-15:
I think I've identified a couple of issues in the installer which are fixed by https://github.com/pimoroni/pirate-audio/commit/0cbcb0014c9becac0c8bbce093266b7721b96f1a

1. If run as `root` it would create `/root/.asoundrc` which would not be available to the "pi" user, thus the missing audio devices
2. If you added `dtoverlay=hifiberrydac` then it's possible the install will conflict since only *one* of these overlays can be active at any one time. This explains the last-overlay-wins fix mentioned by @PelzKo

wagrob, 2021-06-20:
I have bought the microphone-board and landed with the same issue. First "No input device matching mic_out". Then, after fixing this problem with the order in the config.txt (as suggested by @PelzKo ), the clip-recorder now shows "ValueError: No output device matching 'upmix'". No idea how to proceed...

Gadgetoid, 2021-06-22:
@wagrob I have made several changes to the installer here to - hopefully - fix this - https://github.com/pimoroni/pirate-audio/pull/60

wagrob, 2021-06-22:
@Gadgetoid Thanks, the python code is now fixed after I made the recommended changes.

Tuxilop, 2021-09-04:
Same issue with dtoverlay=adau7002-simple and dtoverlay=hifiberry-dac.How using Pirate-udio-mix and another Hat like PHAT DAC or SPEAKER HAT for playing soud ?

closed 2021-09-10 by Gadgetoid