pimoroni/pirate-audio #39: shutil.copyfile kills the dac?

rnsi, 2020-12-27:
I'm using omxplayer with Pirate Audio, works great. But if I run a python script that contains shutil.copyfile, I lose audio. omxplayer still runs as if nothing is wrong, but there's no audio.

The file being copied is a jpg, not the mp3 that's playing. So I can't see how it would affect it. I verified this was the problem by commenting the shutil.copyfile line. Without it, everything works. With that line there's no audio.

Any suggestions?

rnsi, 2020-12-27:
Update-

It's not shutil.copyfile that kills the DAC. It's image.py from st7789-python: https://github.com/pimoroni/st7789-python/blob/master/examples/image.py.

Running that script disables audio. Now what?

duracell80, 2021-01-29:
I reproduced on my headphone board. Good news, I isolated it to the block of code that is assigning pins for the backlight.

I haven't tried iterations of code without those pins but I'm gonna take a look at the overall codebase to see how they are setting up the screen for mopidy.

duracell80, 2021-01-29:
Reading more into the documentation, physical pin 24 is an SPI "chip select" pin noted as being on GPIO8.

CE024 GPIO08 SPI0_CE0_N

rnsi, 2021-02-01:
I changed the backlight pin to 13 and it seems to be working now.

closed 2021-02-01 by rnsi