pimoroni/pirate-audio #40: How can I write an image to the screen?

DeadpoolDiego, 2021-01-02:
I have looked through the code (I have the small speaker board) but am still unsure how to select an image in the image example code. How do I select an image from files? Does the image need to be a certain format? What can I do?

rnsi, 2021-01-02:
https://github.com/pimoroni/st7789-python/blob/master/examples/image.py

python image.py /path/to/image/file.jpg

DeadpoolDiego, 2021-01-02:
Ok I'm kind of a noob. Do I run the script using the Pi's console and put the path to the image after typing image.py or do I have to edit the script in some way.

rnsi, 2021-01-02:
Right. From the console, and put the image after image.py

DeadpoolDiego, 2021-01-02:
Is there a way I can get it to do this automatically when the pi powers on?

rnsi, 2021-01-02:
https://learn.sparkfun.com/tutorials/how-to-run-a-raspberry-pi-program-on-startup/all

DeadpoolDiego, 2021-01-03:
If I use crontab -e to allow the pi to run a script from startup I have to type below it: "@reboot python /path/to/script.py &". Where do I put the path to the image in this command?

rnsi, 2021-01-03:
Just put it in /etc/rc.local

/path/to/python /path/to/script.py /path/to/image &

Find your path to python with:
which python

DeadpoolDiego, 2021-01-04:
There is an error when running the script regularly. It says "/use/local/lib/python2.7/dist-packages/ST7789/_.init.py", line 124, in _init_" No such file or directory. I did also install the st7789 library. How do I fix this?

duracell80, 2021-01-29:
`sudo apt-get update 
sudo apt-get install python-rpi.gpio python-spidev python-pip python-pil python-numpy

sudo pip install st7789`

closed 2023-01-20 by Gadgetoid