pimoroni/pirate-audio #41: Pirate Audio / Raspberry Pi 0 with mopidy, raspotify, upmpdcli

semhoun, 2021-01-12:
Hello,

I have some difficulties to make Pirate Audio working on Raspberry Pi 0 with mopidy, raspotify, upmpdcli, samba and usbmount.

I do a script to install/reinstall my devices, and share it maybe it can help someone: https://gist.github.com/semhoun/926334fcc56fea82d988851c5a6e53ac

snarflakes, 2021-01-27:
does your script work (enable all functionality for all the software?) mopidy was a mess the first time i tried it and I'm kind of stuck. i can't even play mp3.  on the pirate audio, sounds comes out all crackly.1

semhoun, 2021-01-28:
> does your script work (enable all functionality for all the software?) mopidy was a mess the first time i tried it and I'm kind of stuck. i can't even play mp3. on the pirate audio, sounds comes out all crackly.1

I try it on 2 different kit. The only bug I have is on the /etc/asound.conf conf who I have to apply two time on one kit before having PCM mixer running (maybe I forgot an empty line at the end of the file).

snarflakes, 2021-01-29:
There is some conflict in mine, mopidy only works if i kill process APLAY every time I reboot or restart. So i am looking for someone elses.  I have a raspberry pi zero wh.

closed 2023-01-20 by Gadgetoid