pimoroni/pirate-audio #43: Y button on Pirate Headphone generates runtime error

brian-burton, 2021-01-19:
Hi, I'm running a standard script-installed Pirate Audio Headphone attached to a Pi 0 W. I've added dleyna and local mopidy addins, but otherwise I've changed nothing.

I've found that sometimes the Y button fails to work, meaning I can't turn the volume up via buttons (A, B, X all fine), but turning volume up in IRIS is fine.

Syslog shows the following error:

```
Jan 19 17:30:41 pirate-audio mopidy[487]: INFO     [MainThread] mopidy.commands Starting Mopidy core
Jan 19 17:30:41 pirate-audio mopidy[487]: INFO     [MainThread] mopidy.commands Starting Mopidy frontends: RaspberryGPIOFrontend, PiDiFrontend, IrisFrontend, HttpFrontend
Jan 19 17:30:42 pirate-audio mopidy[487]: ERROR    [MainThread] mopidy.commands Got un-handled exception from RaspberryGPIOFrontend
Jan 19 17:30:42 pirate-audio mopidy[487]: Traceback (most recent call last):
Jan 19 17:30:42 pirate-audio mopidy[487]:   File "/usr/lib/python3/dist-packages/mopidy/commands.py", line 241, in _actor_error_handling
Jan 19 17:30:42 pirate-audio mopidy[487]:     yield
Jan 19 17:30:42 pirate-audio mopidy[487]:   File "/usr/lib/python3/dist-packages/mopidy/commands.py", line 438, in start_frontends
Jan 19 17:30:42 pirate-audio mopidy[487]:     frontend_class.start(config=config, core=core)
Jan 19 17:30:42 pirate-audio mopidy[487]:   File "/usr/lib/python3/dist-packages/pykka/_actor.py", line 91, in start
Jan 19 17:30:42 pirate-audio mopidy[487]:     obj = cls(*args, **kwargs)
Jan 19 17:30:42 pirate-audio mopidy[487]:   File "/usr/local/lib/python3.7/dist-packages/mopidy_raspberry_gpio/frontend.py", line 42, in __init__
Jan 19 17:30:42 pirate-audio mopidy[487]:     bouncetime=settings.bouncetime,
Jan 19 17:30:42 pirate-audio mopidy[487]: RuntimeError: Failed to add edge detection
Jan 19 17:30:42 pirate-audio mopidy[487]: INFO     [IrisFrontend-10] mopidy_iris.core Starting Iris 3.55.4
Jan 19 17:30:42 pirate-audio mopidy[487]: INFO     [HttpFrontend-12] mopidy.http.actor HTTP server running at [::ffff:0.0.0.0]:6680
Jan 19 17:30:42 pirate-audio mopidy[487]: INFO     [MainThread] mopidy.commands Starting GLib mainloop
```

Happy to do more debugging.

TrollDecker, 2021-02-10:
Try this:

Edit `/etc/mopidy/mopidy.conf` and change either `bcm20`or `bcm24` to blank, with just one space after the `=`. Don't delete that line or comment it out, because when your configuration is merged with the defaults, it'll include the default setting of that pin. Leaving it blank in your configuration prevents that.

If your board is an original revision, blank `bcm24`. If it's the newer revision, it'll be `bcm20` you'll want to blank.

I _think_ this just solved it for me, but I'm keeping an eye on mine for now.

TrollDecker, 2021-02-10:
Apparently, it did not. 😭

brian-burton, 2021-02-12:
I've done a bit more digging and realised that I raised this against the wrong repo and pimoroni/mopidy-raspberry-gpio#9 already exists.

closed 2021-02-12 by brian-burton