pimoroni/pirate-audio #53: Wrong with scan

ZHUOJIAGAO, 2021-04-15:
When I run the scan command, it will display unrecognized command: local.

[image]

kokosowy, 2021-04-15:
Hi,

I believe you should use mopidyctl command instead:
sudo mopidyctl local scan

If it says local is unavailable check with pip whether mopidy-local is installed.

Gadgetoid, 2021-04-26:
Is this mentioned anywhere in our tutorials/guides? Can't tell if this is an us issue or a Mopidy issue.

kokosowy, 2021-04-26:
It's rather human mistake -- see 'Scanning local media' chapter of https://learn.pimoroni.com/tutorial/sandyj/getting-started-with-pirate-audio.

closed 2023-01-20 by Gadgetoid