pimoroni/pirate-audio #55: modipy.conf duplicate section error?

treatmesubj, 2021-04-26:
Hi there, I've gone through the instructions of setting up Pirate Audio Modipy. I of course, first ran pirate-audio/mopidy/install.sh and rebooted, but the url for the server port never appeared on the screen. I played around with some of the python examples and buttons and screen all seem to make sense.

I then tried the manual setup: https://github.com/pimoroni/pirate-audio/tree/master/mopidy

Near the end, I enabled and started the modipy service. Then, I ran `sudo systemctl status mopidy` to check out what seemed to be going wrong. It appears to indicate that there's a duplicate section in my modipy.conf file, however I'm fairly certain there's not - I just copied over the conf text from the instructions. I'm not sure if it's related to `/boot/config.txt`

Here's the return from `sudo systemctl status modipy`:
```
pi@raspberrypi:/etc/modipy $ sudo systemctl status mopidy
● mopidy.service - Mopidy music server
   Loaded: loaded (/lib/systemd/system/mopidy.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sun 2021-04-25 20:31:29 CDT; 5s ago
  Process: 5751 ExecStartPre=/bin/mkdir -p /var/cache/mopidy (code=exited, status=0/SUCCESS)
  Process: 5752 ExecStartPre=/bin/chown mopidy:audio /var/cache/mopidy (code=exited, status=0/SUCCESS)
  Process: 5753 ExecStart=/usr/bin/mopidy --config /usr/share/mopidy/conf.d:/etc/mopidy/mopidy.conf (code=exited, status=1/FAILURE)
 Main PID: 5753 (code=exited, status=1/FAILURE)

Apr 25 20:31:29 raspberrypi mopidy[5753]:     _load_file(parser, f.resolve())
Apr 25 20:31:29 raspberrypi mopidy[5753]:   File "/usr/lib/python3/dist-packages/mopidy/config/__init__.py", line 194, in _load_file
Apr 25 20:31:29 raspberrypi mopidy[5753]:     parser.read_file(fh)
Apr 25 20:31:29 raspberrypi mopidy[5753]:   File "/usr/lib/python3.7/configparser.py", line 717, in read_file
Apr 25 20:31:29 raspberrypi mopidy[5753]:     self._read(f, source)
Apr 25 20:31:29 raspberrypi mopidy[5753]:   File "/usr/lib/python3.7/configparser.py", line 1065, in _read
Apr 25 20:31:29 raspberrypi mopidy[5753]:     lineno)
Apr 25 20:31:29 raspberrypi mopidy[5753]: configparser.DuplicateSectionError: While reading from '/etc/mopidy/mopidy.conf' [line 56]:
Apr 25 20:31:29 raspberrypi systemd[1]: mopidy.service: Main process exited, code=exited, status=1/FAILURE
Apr 25 20:31:29 raspberrypi systemd[1]: mopidy.service: Failed with result 'exit-code'.
```

Here's the contents of `/etc/modipy/modipy.conf`:
```
pi@raspberrypi:/etc/modipy $ cat /etc/modipy/modipy.conf
[raspberry-gpio]
enabled = true
bcm5 = play_pause,active_low,150
bcm6 = volume_down,active_low,150
bcm16 = next,active_low,150
bcm20 = volume_up,active_low,150

[pidi]
enabled = true
display = st7789

[mpd]
hostname = 0.0.0.0

[http]
hostname = 0.0.0.0

[audio]
mixer_volume = 40
output = alsasink
```

And I guess, here's the contents of `/boot/config.txt`:

```
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
# dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
# dtoverlay=vc4-fkms-v3d
# max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d

# config audio for star wars
#hdmi_drive=2
#hdmi_force_hotplug=1
#hdmi_force_edid_audio=1

# pirate audio stuff
gpio=25=op,dh
dtoverlay=hifiberry-dac
```

Gadgetoid, 2021-04-26:
Not really sure what's happening here, I haven't touched Mopidy in *checks git history* ... 9 months :grimacing: 

Check for files in `/usr/share/mopidy/conf.d` which your systemctl output suggests Mopidy is also using for configuration. I'm not sure, but perhaps a duplicate section in one of those files could cause problems?

That said your debug output does *specifically* mention `/etc/mopidy/mopidy.conf` and claim the duplicate is on line 56... which is quite an accomplishment if the config is indeed only 20 lines long...

I just set up on a relatively fresh Raspbian Buster with our installer and it worked after a reboot. I'll image a new SD card and try again.

treatmesubj, 2021-04-26:
Yeesh, I had created `modipy.conf` and was looking at that, but now that I've had some coffee, I see that it's `mopidy.conf` that I should be working with. I deleted `modipy.conf`.

Now that I listed files in the directory and figured out how to spell, sure enough, my `mopidy.conf` file has lots of duplicates. 

```
pi@raspberrypi:/etc/mopidy $ sudo cat mopidy.conf
# For information about configuration values that can be set in this file see:
#
#   https://docs.mopidy.com/en/latest/config/
#
# Run `sudo mopidyctl config` to see the current effective config, based on
# both defaults and this configuration file.

[raspberry-gpio]
enabled = true
bcm5 = play_pause,active_low,250
bcm6 = volume_down,active_low,250
bcm16 = next,active_low,250
bcm20 = volume_up,active_low,250
bcm24 = volume_up,active_low,250

[file]
enabled = true
media_dirs = /home/pi/Music
show_dotfiles = false
excluded_file_extensions =
  .directory
  .html
  .jpeg
  .jpg
  .log
  .nfo
  .pdf
  .png
  .txt
  .zip
follow_symlinks = false
metadata_timeout = 1000

[pidi]
enabled = true
display = st7789
rotation = 90

[mpd]
hostname = 0.0.0.0

[http]
hostname = 0.0.0.0

[audio]
mixer_volume = 40
output = alsasink device=hw:sndrpihifiberry

[spotify]
enabled = false
username =
password =
client_id =
client_secret =

[raspberry-gpio]
enabled = true
bcm5 = play_pause,active_low,250
bcm6 = volume_down,active_low,250
bcm16 = next,active_low,250
bcm20 = volume_up,active_low,250
bcm24 = volume_up,active_low,250

[file]
enabled = true
media_dirs = /home/pi/Music
show_dotfiles = false
excluded_file_extensions =
  .directory
  .html
  .jpeg
  .jpg
  .log
  .nfo
  .pdf
  .png
  .txt
  .zip
follow_symlinks = false
metadata_timeout = 1000

[pidi]
enabled = true
display = st7789
rotation = 90

[mpd]
hostname = 0.0.0.0

[http]
hostname = 0.0.0.0

[audio]
mixer_volume = 40
output = alsasink device=hw:sndrpihifiberry

[spotify]
enabled = false
username =
password =
client_id =
client_secret =

[raspberry-gpio]
enabled = true
bcm5 = play_pause,active_low,250
bcm6 = volume_down,active_low,250
bcm16 = next,active_low,250
bcm20 = volume_up,active_low,250
bcm24 = volume_up,active_low,250

[file]
enabled = true
media_dirs = /home/pi/Music
show_dotfiles = false
excluded_file_extensions =
  .directory
  .html
  .jpeg
  .jpg
  .log
  .nfo
  .pdf
  .png
  .txt
  .zip
follow_symlinks = false
metadata_timeout = 1000

[pidi]
enabled = true
display = st7789
rotation = 90

[mpd]
hostname = 0.0.0.0

[http]
hostname = 0.0.0.0

[audio]
mixer_volume = 40
output = alsasink device=hw:sndrpihifiberry

[spotify]
enabled = false
username =
password =
client_id =
client_secret =
``` 

Also, I did check out `/usr/share/mopidy/conf.d/mopidy.conf`:
```
pi@raspberrypi:/usr/share/mopidy/conf.d $ cat mopidy.conf
# These are default configurations modified by the mopidy Debian package.
#
# This file should not be modified. Everything can be overriden by
# `/etc/mopidy/mopidy.conf`.

[core]
cache_dir = /var/cache/mopidy
config_dir = /etc/mopidy
data_dir = /var/lib/mopidy

[logging]
format = %(levelname)-8s [%(threadName)s] %(name)s %(message)s
color = false
```

I deleted all the old conf files and created a new one. 

After a reboot, all seems to be well! This time I didn't have to enable and start the mopidy service, it was already running on boot! 

Though the url with the port never did appear on the screen again, the screen shows the volume bar and pause & play icons.

A quick run of `sudo systemctl status mopidy` shows me that the frontend is on port 6680 I believe.

```
pi@raspberrypi:~ $ sudo systemctl status mopidy
● mopidy.service - Mopidy music server
   Loaded: loaded (/lib/systemd/system/mopidy.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2021-04-26 08:20:36 CDT; 1min 42s ago
  Process: 484 ExecStartPre=/bin/mkdir -p /var/cache/mopidy (code=exited, status=0/SUCCESS)
  Process: 490 ExecStartPre=/bin/chown mopidy:audio /var/cache/mopidy (code=exited, status=0/SUCCESS)
 Main PID: 493 (mopidy)
    Tasks: 15 (limit: 4323)
   CGroup: /system.slice/mopidy.service
           └─493 /usr/bin/python3 /usr/bin/mopidy --config /usr/share/mopidy/conf.d:/etc/mopidy/mopidy.conf

Apr 26 08:20:43 raspberrypi mopidy[493]: INFO     [MainThread] mopidy.commands Starting Mopidy mixer: SoftwareMixer
Apr 26 08:20:43 raspberrypi mopidy[493]: INFO     [MainThread] mopidy.commands Mixer volume set to 40
Apr 26 08:20:43 raspberrypi mopidy[493]: INFO     [MainThread] mopidy.commands Starting Mopidy audio
Apr 26 08:20:43 raspberrypi mopidy[493]: INFO     [MainThread] mopidy.commands Starting Mopidy backends: FileBackend, M3UBackend, Str
Apr 26 08:20:43 raspberrypi mopidy[493]: INFO     [MainThread] mopidy.commands Starting Mopidy core
Apr 26 08:20:43 raspberrypi mopidy[493]: INFO     [MainThread] mopidy.commands Starting Mopidy frontends: RaspberryGPIOFrontend, PiDi
Apr 26 08:20:43 raspberrypi mopidy[493]: INFO     [Audio-2] mopidy.audio.actor Audio output set to "alsasink"
Apr 26 08:20:43 raspberrypi mopidy[493]: INFO     [IrisFrontend-9] mopidy_iris.core Starting Iris 3.57.6
Apr 26 08:20:43 raspberrypi mopidy[493]: INFO     [HttpFrontend-11] mopidy.http.actor HTTP server running at [::ffff:0.0.0.0]:6680
Apr 26 08:20:43 raspberrypi mopidy[493]: INFO     [MainThread] mopidy.commands Starting GLib mainloop
```

treatmesubj, 2021-04-26:
actually, idk, I'm not sure how to pull up that frontend in my browser. I ran an ifconfig on my pi and it said my ip was `10.0.0.50` so on my browser I tried visiting `http://10.0.0.50/6680` but to no avail. I'm just not super knowledgeable about networking. I ran an `nmap -Pn 10.0.0.50` but it showed me only port 22 for ssh - I thought maybe I'd see that port 6680 for the frontend.

any ideas?

Gadgetoid, 2021-04-26:
I don't seem to be seeing the URL with the port on the screen, but I *can* access the front end.

Your URL will be something like: http://10.0.0.50:6680/

treatmesubj, 2021-04-26:
Yeah, that's what I figured, and now that I revisit http://10.0.0.50:6680/ the frontend is there. Maybe it takes a min to spin up or something.

Thank you for the help!!

closed 2021-04-26 by treatmesubj