pimoroni/pirate-audio #56: LCD to simply display local time.

GOTO-GOSUB, 2021-04-27:
I am planning on using a Pirate Audio hat to make an Internet radio to replace an existing DAB radio but the killer must-have feature is that it must display the time. I would be quite happy if the LCD just displayed the time but if it could display the date and / or the name of the radio station being played as well that would be a bonus.

But really, I would just be happy for the time. Dead simple - white centered on a black background preferably with fixed width fonts so the placement doesn't jump around when the time updates. Since the Pi obtains the time via NTP shortly after booting it "sounds easy" but I have no experience in making plug-ins or or customising existing code.

Can anyone help in producing a simple display for the time on the Pirate Audio LCD ?

utf18, 2021-05-02:
you can run this in your terminal with python3 

save the code below as time.py and run it with python3 time.py

it will display a white display with the current time, mostly centered with refresh in under a sec

`#!/usr/bin/env python

import time
from colorsys import hsv_to_rgb
from PIL import Image, ImageDraw, ImageFont, ImageColor
from ST7789 import ST7789

SPI_SPEED_MHZ = 80

font = ImageFont.truetype("/usr/share/fonts/truetype/dejavu/DejaVuSans-Bold.ttf", 40)

st7789 = ST7789(
    rotation=90,  # Needed to display the right way up on Pirate Audio
    port=0,       # SPI port
    cs=1,         # SPI port Chip-select channel
    dc=9,         # BCM pin used for data/command
    backlight=13,
    spi_speed_hz=SPI_SPEED_MHZ * 1000 * 1000
)

while True:
    image = Image.new("RGB", (240, 240), (255, 255, 255))
    draw = ImageDraw.Draw(image)
    draw.text((20, 100), time.strftime("%H:%M:%S"), font=font, fill=(0, 0, 0))
    st7789.display(image)

    time.sleep(1.0 / 30)`

GOTO-GOSUB, 2021-05-03:
That is spectacular, @utf18 - thank you for this. I will give that a go shortly.

utf18, 2021-05-03:
we have to find a way to include all this in your aproach for the playlist thing

GOTO-GOSUB, 2021-05-04:
Speaking strictly personally here I would be happy with it simply displaying the time, but saying that I wonder if that feature is something other people might be interested in. It could be that I am a bit weird.

Thinking about this during my commute in to work this morning I was wondering how often people look at the text relating to the current media aspect of the display. In particular I wondered if it would be a nice feature to have the option to say display the media information when this changes but after a certain times (or rotations if this information currently scrolls repeatedly across the display) have it display the time instead.

I can imagine this being set by an entry in a config file where there could be an option to change the behaviour of this text. Ideally we should assume for backwards compatibility that the default setting (or absence of an entry in the config file) is for precisely what it does now.

Then the next option - ie least disruptive to the casual user who is generally happy with how it is now - would be to have a blend of data and time, say the media information is displayed for say 20s (or perhaps 2 or 3 cycles if it scrolls) on a change to the media before displaying the time in a static manner until the next event.

Finally, the next option would be to just display the time on the line currently occupying the media information. 

I say that because I can imagine that the artwork being displayed on the screen is probably of (visual) interest to many and people may not wish to have a blank screen with just the time on it. I quite like the idea of the embedded artwork still being displayed on the LCD as a visual clue to the content being played plus it looks nice.

I should point out that I am speculating as to how it works at present as I ordered the Pirate Audio hat and other bits to make my radio over the weekend and probably will not have the time to look at it until the weekend. This is just how I would probably do it if I knew anything at all about what I was doing !

utf18, 2021-05-07:
sounds like a lot of work, would needed to go into changing the behaviour of this.
i am not using the music feature, since i only use the components, so i am sorry but i am not going to implement all this and not using it myself :)

But maybe the pirate audio team wants to comment on this, and maybe get it in the default bahaviour.

good luck on that ;)

GOTO-GOSUB, 2021-05-07:
That's quite alright - changes to user interfaces are not to be taken lightly and shouldn't inconvenience those already using it. I still haven't forgiven Google for moving the time on my phone from one side to the other !  :)

I will give the python code for the time a try shortly (hopefully in the next few days). Thank you again for this. The physical hardware arrived yesterday so I am hoping to give it a go over the weekend. I really must try to learn Python some day.

It might be interesting to see if anyone else has an interest in the time being displayed as there simply may not be the demand for it.

closed 2023-01-20 by Gadgetoid