pimoroni/pirate-audio #58: spidev 3.4 is fault - free(): invalid pointer

kokosowy, 2021-05-26:
Hi @Gadgetoid!

Your mopidy install script installs spidev package with v3.4, it causes stability issues with Mopidy. It's required to update spidev via pip to version 3.5. Can you please adjust your script to ensure spidev won't be installed with faulty 3.4 version?

Thanks,
Konrad

Gadgetoid, 2021-05-26:
It installs `python3-spidev`, so it's whatever version is supplied by the distro.  I was sure I poked for the package to be updated in Raspberry Pi OS, but it turns out I never actually got 'round to requesting an update (this was happening when the whole world turned upsidedown so I'm not surprised). I have done so now.

Since the installer uses only apt packages I'm remiss to try pulling in anything from pypi- when it comes to packages with C extensions it's... a minefield.

kokosowy, 2021-05-26:
Completely understood. Thanks for looking at this. I mailed Serge, the package maintainer, this morning too.

Gadgetoid, 2021-05-26:
Ah! In that case he'll have two emails pointing this out :laughing: Good shout.

Gadgetoid, 2021-06-15:
It's apparently released and good to go!