pimoroni/pirate-audio #59: Enhancement: turn pidi in a remote control for a remote mopidy server

FelixDefrance, 2021-05-28:
The way is to be able to use Pirate Audio LCD board as a remote control for an existing mopidy setup over a LAN.

Is any devel exist ? It should be great !

Gadgetoid, 2021-09-10:
In theory PiDi supports this, but I can't for the life of me remember how you get it up and running... too many projects! - https://github.com/pimoroni/pidi

PiDi is basically the standalone version of the Mopidy plugin.

Gadgetoid, 2021-09-10:
Turns out PiDi is a fairly neglected, broken mess and I've done some work to get it back up and limping, but it's not ideal.

From what I remember, it doesn't support any button control but that could be accomplished with another script like this -  https://github.com/pimoroni/pirate-audio/blob/master/examples/shairport-sync-control.py - except using MPDClient and sending the relevant play/stop/vol-up/vol-down commands.

closed 2023-01-20 by Gadgetoid