pimoroni/pirate-audio #65: Clarification for the Setup instruction needed

nephx, 2021-09-10:
Hi there,
I just spend some hours trying to get my Raspy Zero / Pirate Audio to scan local media and just feel like a complete noob because I didn't understand, that I had to put the [local] section with media_dir = /home/pi/Music in the config file and instead edited media_dirs. Perhaps 
> Change the line that says media_dir = /var/lib/mopidy/media in the [local] section so that it reads media_dir = /home/pi/Music

could be changed to 
> Add a [local] section and put media_dir = /home/pi/Music in it

Then there's the problem that the instruction ends exactly where the interesting part begins. 
I learn how to remote control the pirate audio from a computer, but how do I use it as a stand-alone player? Buttons don't work, so what do I have to do to control it directly?

I'm kinda lost here, so any help would be appreciated.

Gadgetoid, 2021-09-10:
I'll look into clarifying the setup instructions when I get time.

As for a standalone player- Mopidy + Pirate Display is not and will not ever be a complete stand alone player. It's just too complicated to write a full LCD front-end for Mopidy that enables song browsing, etc. It just gives transport + volume control for playlists you've already started playing through the web UI.

Pirate mp3 sort-of functions as a standalone mp3 player, but I haven't looked at this code for a while and can't remember where I got with it - https://github.com/Gadgetoid/pirate-mp3

nephx, 2021-09-10:
Yes, I learned that in the meantime and I have to say that I would call things like

    "Build your own home-brew iPod Nano with Pirate Audio Headphone Amp! It's an all-in-one solution, with gorgeous album art
    display, track info, and playback controls, for playing your local audio files (MP3, FLAC, etc) or streaming music from Spotify."

pretty misleading, and bought the Pirate Audio under completely false assumptions. Still a nice toy, but if I need  a device to control most functions of an audio player, why don't I use the device directly to play audio.

I'll go and check-out pirate-mp3

Nevertheless, the problem is, that remote controlling works, but none of the buttons do anything, so I can't even pause or change the volume on the Pirate Audio.

Gadgetoid, 2021-09-13:
I'd tend to agree that it's possibly a little misleading, and have raised it internally for some clarification. While we don't explicitly say full playlist browsing, invoking the name "iPod Nano" kinda implies it.

Regarding GPIO input, make sure the `mopidy-raspberry-gpio` plugin is installed and enabled.

If you run `systemctl status mopidy` you should see a line something like:

```
Sep 13 09:47:53 bullseye mopidy[8075]: INFO     [MainThread] mopidy.commands Starting Mopidy frontends: RaspberryGPIOFrontend
```

which should include: `RaspberryGPIOFrontend`

closed 2023-01-20 by Gadgetoid