pimoroni/pirate-audio #72: Can the script be made available for for my Raspberry zero pi running Wheezy?

TheRealBrommer, 2021-12-04:
I have a raspberry pi zero running with my own Thingspeak & Owncloud server. It's running WHEEZY as that was the only way i could make the older ThingSpeak server working (a lot of hurdles, but i got there and works great). 

I'd like to add even more functionality to it by using the Pimoroni Pirate-audio lineout to out, but the install script is coming off the rails a lot (unable to find things, comes up a lot). I know i could create a dedicated zero pi for this (with better OS compatibility), however those things seem to be sold out worldwide + the load on the existing zero pi is low.

Just wondering if anyone has already done this or someone with more python knowledge managed to get it working on WHEEZY.

TheRealBrommer, 2022-01-19:
Nevermind guys. No help at all hey, not even a peep.

Forget about using this install script. Iris as an interface is horrible anyway to say the least. 

Solution? Install pirateAudio on Volumio and use the wonderful plugin developed by AxLED. It's absolute fantastic, unlike what's provided here. AND he provides help, you can find it here:

https://community.volumio.org/t/plugin-pirate-audio/44336

Gadgetoid, 2022-01-20:
You could have posted your update without throwing quite so much shade :roll_eyes: 

But, yeah, if people want to use something else... great! There's only one of me stretched micron-thin between hundreds of products and thus there are reasonable limits to what I can do. Supporting seven year old operating systems for one persons weird and highly specific setup takes something of a low priority.

closed 2022-01-20 by Gadgetoid

TheRealBrommer, 2022-01-21:
I didn't realise it was a lone-ranger operation. 

So point taken.

Sorry.

Gadgetoid, 2022-01-22:
> I didn't realise it was a lone-ranger operation.

To be fair the *company* is far more than one person so it's easy to assume we've got a big ol' team working on these things. In reality we've only just brought aboard a second software engineer. I'm spinning more plates than I can handle here, so you're coming back and posting your findings (and even helping out in other Issues) is appreciated.

Hiring slow and steady so we can weather a crises (of which there have been... quite a few lately) has its downfalls :grimacing: - I certainly don't enjoy letting the ball drop like this :disappointed:

hartogsmith, 2022-01-22:
> been... quite a few lately) has its downfalls 😬 - I certainly don't

sounds like the real issue is selling hardware that's built a very specific
way without having the staff to provide enough support for the hardware.

Gadgetoid, 2022-01-22:
We *do* have support staff, this is just the wrong place to reach them. GitHub doesn't really provide very good tools to communicate that.

Most large software operations have the benefit of all the code being private, the developers unreachable, and a 100-fold team of people reading from call-center scripts to give the illusion of support. Us? Not so much!