pimoroni/pirate-audio #73: OLED screen flickers when playing music

TheRealBrommer, 2021-12-05:
The script works fin on Raspberry PI OS Bullseye. 

HOWEVER, when i play music (with mopidy-tuneIn; i also have the spotify one installed but can't get than one to work yet: do you really need a premium account for that?) the oled lcd flickers ! 

When i pause the music i get a steady screen is there any reason for this?

TheRealBrommer, 2021-12-05:
I suspect myself it might have something to do with power (currently i'm running the whole thing off a usb port of my router which is what ? 500 mA ?)

TheRealBrommer, 2021-12-10:
Just tried a 2A power plug - same issue. INCREDIBLY annoying......

TheRealBrommer, 2021-12-10:
Interesting forum here where people have to solve their own problem.........

anyway flickering is SOLVED. Yes i know exactly what fixed it. I feel tempted to keep it to myself, because of all the help i received here.....

I'm too nice for that. OK so here goes. My hat is connected by wires, as i have other things connected as well. So changes are you might forget one. Also not that the pinout diagram on xyz is not complete ! So:
1. make sure you connect pin 17 the 3.3V as well
2.  you can't stop the timeout *screen going black" after a certain time by adding asome additional code to the mopidy.conf under [pidi]. Add "idle_timeout = 60000" (that's what i have and i'm assuming this is seconds; make it big and should last all day pretty much). Reference is here: https://github.com/pimoroni/mopidy-pidi/blob/master/mopidy_pidi/ext.conf

Gadgetoid, 2022-01-20:
> Interesting forum here where people have to solve their own problem.........

This is not a forum. It's supposed to be a bug tracker. Your first port of call should be the Discord, forums or support email. I'm the only one watching GitHub Issues and I tend to be mired in some software project or another, so responses and turnarounds can be SLOW here.

Thanks for posting your fix, however, and directing others around the display sleep pitfall. That probably needs a documentation or default fix.

(You can indicate that an issue is resolved by closing it. That runs counter to normal forum etiquette because Issues are ostensibly a TODO list for me personally, and not a place for open discussion)

closed 2022-01-20 by Gadgetoid