pimoroni/pirate-audio #76: LCD Display stops working after some time....

Platypuschan, 2022-01-16:
LCD Display stops working after some time, music still playing, mopidy still running, buttons still working, backlight of screen lights up while song changes/key-presses. Restarting the whole mopidy daemon solves the problem...
Is there a way just to restart the mopidy-display-program?

TheRealBrommer, 2022-01-19:
I had this too. I bet you're seeing what i saw, a timeout. I mention it here in the last post and how i fixed it: 

https://github.com/pimoroni/pirate-audio/issues/73

maybe that'll solve it for you.

Gadgetoid, 2022-01-20:
If what @TheRealBrommer suggests turns out to fix it for you, we probably need a tweak to the default value or documentation to make this clearer.

Platypuschan, 2022-08-04:
I don´t think it´s about the power-consumption... The screens backlight is still on, music is still playing and stuff. Just the screen is shut off, usually you can´t connect with the mopidy-android-app until mopidy is restarted and bogus like that. I think mopidy get´s stuck and needs to be restarted.

closed 2022-08-04 by Platypuschan