pimoroni/pirate-audio #88: NO_PUBKEY?

mundabiddi, 2023-04-24:
Sir,
apparently the installer needs updating,.
Amongst other errors when following the Pirate Audio instructions on the Pimoroni web site I get:-
The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 78FD980E271D2943

Help please....
Thanks!

winitalia, 2023-04-29:
Hi @mundabiddi 
Try these commands first and then restart the installation.

`sudo mkdir -p /etc/apt/keyrings`
`sudo wget -q -O /etc/apt/keyrings/mopidy-archive-keyring.gpg \
  https://apt.mopidy.com/mopidy.gpg`
`sudo wget -q -O /etc/apt/sources.list.d/mopidy.list https://apt.mopidy.com/bullseye.list`

mundabiddi, 2023-05-01:
Sorry, still not working!

This is the first part of the installation which shows the NO_PUBKEY errors.

Regards,

Michael.

***@***.***:~ $ sudo mkdir -p /etc/apt/keyrings

***@***.***:~ $ sudo wget -q -O 
/etc/apt/keyrings/mopidy-archive-keyring.gpg \ 
https://apt.mopidy.com/mopidy.gpg

***@***.***:~ $ sudo wget -q -O 
/etc/apt/sources.list.d/mopidy.list https://apt.mopidy.com/bullseye.list

***@***.***:~ $ git clone https://github.com/pimoroni/pirate-audio

Cloning into 'pirate-audio'...

remote: Enumerating objects: 349, done.

remote: Counting objects: 100% (63/63), done.

remote: Compressing objects: 100% (30/30), done.

remote: Total 349 (delta 38), reused 36 (delta 32), pack-reused 286

Receiving objects: 100% (349/349), 138.18 KiB | 516.00 KiB/s, done.

Resolving deltas: 100% (156/156), done.

***@***.***:~ $ cd pirate-audio/mopidy

***@***.***:~/pirate-audio/mopidy $ sudo ./install.sh

Updating apt and installing dependencies

Get:1 http://raspbian.raspberrypi.org/raspbian bullseye InRelease [15.0 kB]

Hit:2 http://archive.raspberrypi.org/debian bullseye InRelease

Get:3 https://apt.mopidy.com bullseye InRelease [14.1 kB]

Err:3 https://apt.mopidy.com bullseye InRelease

The following signatures couldn't be verified because the public key is 
not available: NO_PUBKEY 78FD980E271D2943

Reading package lists... Done

W: GPG error: https://apt.mopidy.com bullseye InRelease: The following 
signatures couldn't be verified because the public key is not available: 
NO_PUBKEY 78FD980E271D2943

E: The repository 'https://apt.mopidy.com bullseye InRelease' is not signed.

N: Updating from such a repository can't be done securely, and is 
therefore disabled by default.

N: See apt-secure(8) manpage for repository creation and user 
configuration details.

On 29/04/2023 09:28, winitalia wrote:
> |sudo wget -q -O /etc/apt/sources.list.d/mopidy.list 
> https://apt.mopidy.com/bullseye.list|

helgibbons, 2023-05-10:
The installer should be fixed as of #87 - I found I needed to delete my existing Mopidy sources for it to grab the correct ones though:

`sudo rm -r /etc/apt/sources.list.d/mopidy.list`

helgibbons, 2023-05-18:
Ah yes - I'm getting the same with a fresh install from `main`.

You could try pulling the version from #89 - I think this contains a few more tweaks to the install script?

```
git clone -b patch-bullseye-and-spotify https://github.com/pimoroni/pirate-audio
cd pirate-audio/mopidy
sudo ./install.sh
```

helgibbons, 2023-05-19:
(Just tried this and it still gives me an error about 'NO_PUBKEY', but Mopidy seems to install successfully.)

mundabiddi, 2023-05-19:
Worked for me too!!

(Also with the NO-PUBKEY error mentioned below but I can live with that!!)

Gadgetoid, 2023-05-19:
After playing another game of spot the difference, I found one small error: https://github.com/pimoroni/pirate-audio/pull/91

closed 2025-01-08 by Gadgetoid