pimoroni/pirate-audio #92: button image

Wanaming800, 2023-06-08:
inside **_mopidy.conf_** i changed `A` button function

`bcm5 = play_pause,active_low,150`
to
`bcm5 = prev,active_low,150`

but how to change hint image on display to match new button function?

Gadgetoid, 2023-06-13:
The controls are an overlay image which lives in the `pidi-display-pil` plugin for `mopidy-pidi`. It's a bit of a rabbit hole: https://github.com/pimoroni/pidi-plugins/tree/master/pidi-display-pil/pidi_display_pil/images

You should, in theory, be able to find these on your local filesystem and edit them.

You might be able to locate it with:

```
python3 -c "import importlib;print(importlib.util.find_spec('pidi-display-pil').origin)"
```

Wanaming800, 2023-06-13:
> The controls are an overlay image which lives in the `pidi-display-pil` plugin for `mopidy-pidi`. It's a bit of a rabbit hole: [...]

Thanks for the clue
Found both images and config at  `/usr/local/lib/python3.9/dist-packages/pidi_display_pil/`

closed 2023-06-13 by Wanaming800