pimoroni/pirate-audio #93: After installing then running I get the following error: ValueError: No output device matching 'default'

7sharp9, 2023-06-11:
```
pi@raspberrypi:~/GitHub/pirate-audio/clip-recorder $ python cliprecord.py
Traceback (most recent call last):
  File "/home/davethomas/GitHub/pirate-audio/clip-recorder/cliprecord.py", line 354, in <module>
    recordamajig = Recordamajig()
  File "/home/davethomas/GitHub/pirate-audio/clip-recorder/cliprecord.py", line 82, in __init__
    self._out_stream = sounddevice.OutputStream(
  File "/usr/local/lib/python3.9/dist-packages/sounddevice.py", line 1494, in __init__
    _StreamBase.__init__(self, kind='output', wrap_callback='array',
  File "/usr/local/lib/python3.9/dist-packages/sounddevice.py", line 817, in __init__
    _get_stream_parameters(kind, device, channels, dtype, latency,
  File "/usr/local/lib/python3.9/dist-packages/sounddevice.py", line 2659, in _get_stream_parameters
    device = _get_device_id(device, kind, raise_on_error=True)
  File "/usr/local/lib/python3.9/dist-packages/sounddevice.py", line 2799, in _get_device_id
    raise ValueError(
ValueError: No output device matching 'default'
```

Any ideas on what to check?

```
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

7sharp9, 2023-06-11:
`arecord -Dmic_out -c2 -r48000 -fS32_LE -twav -d5 -R10000 -Vstereo test.wav`
This does indeed work so the HW seems to work its just the `cliprecord.py` does not seem to work correctly.

Gadgetoid, 2023-06-13:
Check `aplay -l` and `aplay -L` these should give you an idea of what output devices you have.

Then try editing this line: https://github.com/pimoroni/pirate-audio/blob/d9d7cadf1686860dcbfe42b74d34454942cba2d4/clip-recorder/cliprecord.py#LL354C30-L354C30

To:

```python
recordamajig = Recordamajig(output_device="your_device")
```

Might be that a Raspberry Pi OS update has changed the device names.

7sharp9, 2023-06-13:
```
pi@raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 1: vc4hdmi [vc4-hdmi], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
 ```

```
pi@raspberrypi:~ $ aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)
lavrate
    Rate Converter Plugin Using Libav/FFmpeg Library
samplerate
    Rate Converter Plugin Using Samplerate Library
speexrate
    Rate Converter Plugin Using Speex Resampler
jack
    JACK Audio Connection Kit
oss
    Open Sound System
pulse
    PulseAudio Sound Server
upmix
    Plugin for channel upmix (4,6,8)
vdownmix
    Plugin for channel downmix (stereo) with a simple spacialization
mic_hw
mic_rt
mic_plug
mic_filter
mic_out
usbstream:CARD=adau7002
    adau7002
    USB Stream Output
hw:CARD=vc4hdmi,DEV=0
    vc4-hdmi, MAI PCM i2s-hifi-0
    Direct hardware device without any conversions
plughw:CARD=vc4hdmi,DEV=0
    vc4-hdmi, MAI PCM i2s-hifi-0
    Hardware device with all software conversions
default:CARD=vc4hdmi
    vc4-hdmi, MAI PCM i2s-hifi-0
    Default Audio Device
sysdefault:CARD=vc4hdmi
    vc4-hdmi, MAI PCM i2s-hifi-0
    Default Audio Device
hdmi:CARD=vc4hdmi,DEV=0
    vc4-hdmi, MAI PCM i2s-hifi-0
    HDMI Audio Output
dmix:CARD=vc4hdmi,DEV=0
    vc4-hdmi, MAI PCM i2s-hifi-0
    Direct sample mixing device
usbstream:CARD=vc4hdmi
    vc4-hdmi
    USB Stream Output
```

`default` is listed with the `-L` but it doesnt find it in the python script, I dont think I have any devices listed that have an actual output?

7sharp9, 2023-06-13:
If I list the devices with `python3 -m sounddevice`
```
pi@raspberrypi:~ $ python3 -m sounddevice
   0 adau7002: bcm2835-i2s-adau7002-hifi adau7002-hifi-0 (hw:0,0), ALSA (2 in, 0 out)
   1 sysdefault, ALSA (128 in, 0 out)
   2 lavrate, ALSA (128 in, 0 out)
   3 samplerate, ALSA (128 in, 0 out)
   4 speexrate, ALSA (128 in, 0 out)
   5 upmix, ALSA (8 in, 0 out)
   6 vdownmix, ALSA (6 in, 0 out)
   7 mic_hw, ALSA (2 in, 0 out)
   8 mic_rt, ALSA (2 in, 0 out)
   9 mic_plug, ALSA (128 in, 0 out)
  10 mic_filter, ALSA (128 in, 0 out)
  11 mic_out, ALSA (128 in, 0 out)
> 12 default, ALSA (128 in, 0 out)
```

ubudev1397, 2024-04-12:
@7sharp9 did you ever figure this out. I am having the same problem