pimoroni/pirate-audio #99: Pirate Audio Dual Mic - Raspberry Pi 5 ?

radames, 2024-01-29:
Hello, I got this beautiful component that I want to make it work with the Raspberry Pi 5.

First, `dtoverlay=adau7002-simple` does not appear to load the appropriate kernel audio module.
Do you know if the new I2C bus remains compatible with the original Pirate Audio Dual Mic design?

radames, 2024-01-29:
solved, by updating to the latest kernel 
```
sudo apt-get update
sudo apt-get dist-upgrade
```
Then edit `config.txt` on the location

```
sudo vim /boot/firmware/config.txt

add 
dtoverlay=adau7002-simple
```
Reboot

closed 2024-01-29 by radames

lukifer, 2024-08-16:
@radames Following up, is audio recording still working for you on the Pi 5? I've run `dist-upgrade` and applied `dtoverlay=adau7002-simple`, and while I see the ALSA device, I can't get it to record using `arecord`, Pulse, or `sounddevice` in Python (it acts like it works, but only records silence; an identical setup works as expected on a Pi 4).

If it's working for you, any chance you could share your `config.txt`? Thanks so much!