pimoroni/pirate-audio #102: Pi 5: Dual-mic audio input is silent

lukifer, 2024-08-16:
I'm using a Dual Mic on a Pi 5, and I was able to get the LCD and buttons working using a Python virtual environment, and replacing `rpi-gpio` with [rpi-lgpio](https://rpi-lgpio.readthedocs.io/en/latest/).

However, I'm unable to get any audio to record (it seems to work, but captures only silence), whether via the `mic_out` virtual source, or using the device directly (`plughw:CARD=adau7002`). I've also tried Pulse, and the `sounddevice` import in the `cliprecorder` demo, and no luck, as well as running everything through `sudo`. A matching ALSA config works as expected on a Pi 4.

Has anyone gotten audio recording working on a Pi 5?

lhypds, 2024-12-21:
I'm using Pi Zero 2W, same issue, anyone can help?

koh43, 2025-09-01:
Change the device of the `Recordamajig` object from "mic_out" to "default".  

You can check which device is set in the sounddevice library by running the following command:  

`python3 -m sounddevice`  

The active device will have a `*` on the left.  

Here’s an example from my Raspberry Pi 5:

```
  0 adau7002: 1f000a0000.i2s-adau7002-hifi adau7002-hifi-0 (hw:2,0), ALSA (2 in, 0 out)
  1 pulse, ALSA (32 in, 32 out)
  2 mic_rt, ALSA (2 in, 0 out)
  3 mic_plug, ALSA (128 in, 0 out)
  4 mic_filter, ALSA (128 in, 0 out)
  5 mic_out, ALSA (128 in, 0 out)
* 6 default, ALSA (32 in, 32 out)
```